OctoPrint/OctoPrint #3410: [FR] During analysis, look for things like {travel_speed}

OutsourcedGuru, 2020-01-01:
**Is your feature request related to a problem? Please describe.**
We keep seeing this `{travel_speed}` appearing in the Get Help section of the forum. It's like this will never go away.

```
Send: N17 G1 F {travel_speed}*126
Recv: ok N:6
Send: N18 M105*30
Recv: //Contain Unknown opts:N17 G1 F {travel_speed}*126
Recv: 
Recv: //sd cmd error cnt:7 len:28 :N17 G1 F {travel_speed}*126
Recv: 
Recv: ok
Recv: Resend:7
Send: N19 M117 Imprimiendo*119
Recv: //last:6 curr:18 rcv:N18 M105*30
Send: N7 G28 Z0*94
Recv: 
Recv: //callback_pf error
Recv: Resend:7
Recv: //last:6 curr:19 rcv:N19 M117 Imprimiendo
Recv: //callback_pf error
```

**Describe the solution you'd like**
At the point where the file is first uploaded and analyzed, it might be nice to search for those french braces and to possibly refuse to allow it to exist or to throw a popup message indicating that it won't work.

**Describe alternatives you've considered**
Even though a slicer-based fix is what's called for, the users still come to the OctoPrint forum for support.

OutsourcedGuru, 2020-01-10:
I note that attempting to add a log entry within the gcodeInterpreter.py file like this:

```
self._logger.warn("GCODE included \{travel_speed\} rather than the expected \{speed_travel\}")
```

...results in a complaint that there's no logger associated with it. It's not really related but presumably there's no logging endpoint for this module.

foosel, 2020-06-22:
I've added a new bundled plugin (on a difference release cycle though): https://github.com/OctoPrint/OctoPrint-FileCheck

That will check uploaded files for common issues. Currently it only checks for `{travel_speed}` in uploaded GCODE files. If something is detected, a notification gets pushed to the clients indicating the problem.

I have decided to not make it part of the GCODE analysis as that might be replaced from a plugin. I've also decided against making it part of the file upload itself for performance reasons.

fieldOfView, 2020-06-23:
For completeness, `{travel_speed}` should be properly replaced with the expected number again since Cura 4.1. You still probably should not use it though, because the number Cura inserts is in mm/s, whereas the firmware expects mm/min AFAIK.

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel