OctoPrint/OctoPrint #3412: [Request] please add a 16:9 screen layout - best auto-switched.

dothebart, 2020-01-02:
Most computer displays nowadays are 16:9 - 4:3 s vanished. Octoprints UI however mostly shows 4:3 layout. The most anoying result of that is that the file upload button and the file selector dialog is out of reach when using the rest of the UI. 
Please use the full available screen realestate for octoprint.

tedder, 2020-01-02:
@dothebart how are your css/responsive skills?

dothebart, 2020-01-03:
Well, if they'd be anywhere near mentionable I'd have sent a pr ;) I'm a C/C++ Backend guy.

LMS0815, 2020-01-04:
Might be possible to add something to:  [Themeify plugin](https://plugins.octoprint.org/plugins/themeify/)

jneilliii, 2020-01-04:
Details on the community forum. 
 https://community.octoprint.org/t/pimp-my-web-interface/3349

Also, the widescreen plugin allows you to micr some things around.

dothebart, 2020-01-06:
Ah, thanks for the sugestions, I was searching the plugins, but didn't find these. 
Can any of these be made the sane default in future?

tsmes, 2020-01-06:
It seems like the UI depends on Bootstrap v2.3.1 released in 2012. I could take a crack at updating the UI to a newer layout but before taking on a task like that it would be nice to know what the target browser support is.

fieldOfView, 2020-01-06:
@tsmes: Not an answer to your question, but nevertheless a discussion with some possible directions here:
https://github.com/foosel/OctoPrint/issues/2420

LMS0815, 2020-01-09:
Themeify plugin / Try this to dynamically use the full screen width:
[image]

`calc(95% - 340px)`

stantond, 2020-01-19:
A slight tangent... maybe this raises the need for a tech debt issue to update to the latest version of Bootstrap (currently v4.4.1 at https://getbootstrap.com/).

@foosel maybe some of the 1.4.x iterations could be updating other components and dependencies before adding features, given it's the Python 3 release? Or maybe it's best to leave 1.4 to stabilise after release, and think of other dependency updates from 1.5? (P.s. let me know if you could use a hand building a roadmap, can't contribute production quality code but can certainly help with product management).

johankvint, 2020-02-08:
@tsmes I would be happy to help.
Updating to bootstrap 4 should not be "that" hard.
And we could also keep the grid classes from old bootstrap to support most plugins.

Main reason I started looking in here was to see if there where any plans for better file management, but this could be a fun project too :)

jasonbcox, 2021-02-09:
I've come across this issue as well. I can look into authoring a patch myself, but the backwards compatibility requirement may negatively impact code quality until the next major release (I imagine lots of inlined if-statements. not good). I'm not sure if this is something the maintainers are ok with, and such an impactful change will likely require closer coordination.

Some thoughts:
* Is there any data to indicate how many plugins (and by extension, users) are impacted specifically by reflowing the UI horizontally?
* How many of the above plugins can be safely eliminated with a better UI? Any "wide layout" type plugins would be no longer necessary, for example.
* Is there a reason to keep Bootstrap at all? Many projects use [flexbox](https://css-tricks.com/snippets/css/a-guide-to-flexbox/) directly. This would eliminate a dependency and flexbox is supported on all modern browsers these days.

cp2004, 2021-02-09:
There is work ongoing to prepare for a UI re-write. For example, creating https://bundleviewer.octoprint.org React and Material UI were used. Plugins like OctoPrint widescreen work with most plugins, so there shouldn't be much of an issue just re-sizing things per se. But it is a decent change to the look and feel that there is little enthusiasm for when a complete redesign is on the horizon.

Backwards compatibility is a huge issue here, such that the plan is to use [UI plugins](https://docs.octoprint.org/en/master/plugins/mixins.html#uiplugin) to allow us to test concepts and create a gradual switch over.

LazeMSS, 2021-03-31:
Possible by plugin:

https://plugins.octoprint.org/plugins/uicustomizer/

closed 2021-03-31 by LazeMSS

jneilliii, 2021-03-31:
Don't forget [Consolidated Tabs](https://plugins.octoprint.org/plugins/consolidatedtabs/) or [Widescreen](https://plugins.octoprint.org/plugins/widescreen/) plugins, there may be more.