OctoPrint/OctoPrint #3415: [Request] Define backup plugin path

ModischFabrications, 2020-01-04:
**Is your feature request related to a problem? Please describe.**
I want to backup my octoprint instance via CLI with the packaged backup plugin and a cron script, but I can't change the path where the backup is stored. This makes further processing difficult and dangerous, as I don't want to access the whole octoprint folder from the backup script.

**Describe the solution you'd like**
I would like the plugin to take a path, similar to the restore functionality. Making this path optional won't introduce breaking changes. This would result in `octoprint plugins backup:backup [OPTIONS] [PATH]`. 

**Describe alternatives you've considered**
Add the default path to the "Folders"-Entry of the web interface. This won't help for a fully CLI-driven setup, but it's the best alternative I can think of.

ModischFabrications, 2020-01-04:
Oh, btw, the default backup path is not mentioned anywhere except for the logs, it might be sensible to add it to the readme or print it in the UI.

foosel, 2020-06-18:
This has been implemented in b399ce6fbcf11879f511dff2e72586470829de12 and will be part of 1.4.1.

Syntax will be `octoprint plugins backup:backup --path <backup file>`

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel