OctoPrint/OctoPrint #3417: Tornado.application error: too many open files

bahmrockk, 2020-01-05:
Hello all,
I think I stumbled across an edge case leak while developing a plugin. Please check if this is truly a bug and if not or if it's not relevant, happily disregard!

#### What happened / symptoms
The frontend became inaccessible, claiming that: "The OctoPrint server is currently not running" (looks like an Octopi error, not octoprint though). Printing on the Ender 3 Pro continued without issues, confirming that octoprint is not a ghost process but really doing work.

#### What were you doing?
* running octoprint for 30h+ with several plugins activated (see below). One of them, my own, spammed the log file on accident.
* Trying to check a print progress in the morning, the webservice returns "OctoPrint server is currently not running". Printer still does though.
* Checking the logs, I realized two things: 1) I forgot a debug message in my test plugin and 2) tornado regularly throws an exception in callback: 
```
tornado.application - ERROR - Exception in callback (<socket._socketobject object at 0x728a5030>, <function null_wrapper at 0x71451ef0>)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/ioloop.py", line 888, in start
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/stack_context.py", line 277, in null_wrapper
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/netutil.py", line 264, in accept_handler
  File "/usr/lib/python2.7/socket.py", line 206, in accept
error: [Errno 24] Too many open files
```

#### Versions & other info:
   Python:        2.7.16 
   OctoPrint     1.4.0rc3 
   OctoPi          0.17
   RPi.bme280 0.2.8 (from the plugin)

#### The Log File
The errors start at line 3249 of the octoprint.log - I didn't truncate even the previous running cycles in case there's some vital information I've missed. The error repeats for another 667093 lines, but those I truncated due to size :)

[octoprint.log](https://gist.github.com/bahmrockk/cce1aad2d914fcdbc0ca8d27209e44e8)

ps aux has three processes related to octo, mainly the octoprint server giving its best and doing work:
```
avahi      365  0.0  0.3   5896  2912 ?        Ss   Jan03   0:06 avahi-daemon: running [octopi.local]
root       423  0.2  0.4  52976  3660 ?        Sl   Jan03   5:11 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0
pi        9921 25.5  8.3 266352 74756 ?        Rl   Jan04 291:48 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000
```

htop confirmed this:
[image]

Thank you for an awesome tool and an awesome community!

Kind regards,
Georg

foosel, 2020-01-13:
Have you ever seen this happen with 1.3.12 as well, or only with 1.4.0?

bahmrockk, 2020-01-13:
Only with 1.4.0, untested with 1.3.x - I switched when starting to look at the dev tutorials.

foosel, 2020-01-13:
Too me this looks like something in the process is eating up all file handles and then there are none left over for Tornado to create a connection (socket connections also need file handles) and hence the webserver can't be accessed anymore.

If this doesn't happen with your plugin disabled/not installed I'd suggest to take a hard look at how you open any kind of files and if you properly close them again (`with open(....) as f:` instead of just `f  = open(....)`), or if the used library is sound in that regard.

Since it doesn't handle in safe mode however according to you, it's definitely a third party plugin that's at fault of this - I can't say if your's or something else installed.

foosel, 2020-01-13:
Closing because it doesn't occur in safe mode, however comments are still enabled :)

closed 2020-01-13 by foosel