OctoPrint/OctoPrint #3419: [Brainstorming] PRINT permission without FILES_LIST

fieldOfView, 2020-01-06:
**Is your feature request related to a problem? Please describe.**
OctoPrint 1.4 introduces granular permissions. The PRINT permission logically includes the FILES_SELECT permission (how else would OctoPrint know what file to print?). The FILES_SELECT permission logically includes the FILES_LIST permission (how else would the user be able to select an existing file?). Thus, any user who can start a print, automatically can also see all gcode files.

As an application developer using the API key, I may want to allow users to start a print they just uploaded through my application, but restrict their access to a list of other files already on the OctoPrint instance. In other words: I want users to be able to PRINT without the FILES_LIST permission.

**Describe the solution you'd like**
I think I would like the FILES_SELECT permission to not automatically inherit the FILES_LIST. This might be a little strange from the perspective of the OctoPrint web interface (because how do you select a file if you cannot see the available files?) but from the standpoint of an application that uploads a gcode file it is not necessary to be able to list all files if the application knows what file it just uploaded. 

**Describe alternatives you've considered**
The only alternative I can think of is to life with a PRINT permission that also includes a FILE_LIST permission, making the permission system less granular.

**Additional context**
Eventually a solution where a user can (optionally) only list (and start, pause, cancel) her own gcode files would be great, but that is a bigger change for a future version of OctoPrint.

foosel, 2020-01-13:
You know, you've actually just made me question the sanity of the design decision overall to include permissions in other permissions (unless those other permissions are admin/management permissions). I'll have to poke a bit at the code and try to jog my memory what the reasoning behind that was, since indeed from a strictly permission modelling point of view, it seems a bit arbitrary and tailored to the core UI.

fieldOfView, 2020-01-13:
> you've actually just made me question the sanity of the design decision

sorry :-/

foosel, 2020-01-14:
> sorry :-/

Why sorry? That is a *good* thing to come up now rather than once 1.4.0 stable is released, so thank you!

I've changed the implementation to not include assumptions about the UI into the permission modelling and so far nothing seems to have broken, so that looks like a sound solution.

foosel, 2020-01-14:
Should be solved by the above commit.

fieldOfView, 2020-01-14:
> Why sorry?

Your sanity being checked, especially by yourself, is never a nice. Not sorry about finding a bug ofcourse :-)

foosel, 2020-01-28:
1.4.0rc4 is out

closed 2020-01-28 by foosel