OctoPrint/OctoPrint #3420: [Request] Adding a Web App Manifest for "Add to Homescreen" (Android Chrome).

JohnEdwa, 2020-01-06:
**Is your feature request related to a problem? Please describe.**
Using Octoprint with a mobile browser has the URL bar and such at the top, obscuring the view, and it is also just another tab. But it could be better.

**Describe the solution you'd like**
Adding a [Web App Manifest](https://developers.google.com/web/fundamentals/web-app-manifest) that includes "display": "standalone" will allow (Android Chrome at least) to properly utilize the "Add to Homescreen" functionality which will create an app shortcut that will then launch Octoprint in a completely UI-less browser window, mimicking a native app. As a bonus, it will register itself to the app-switcher too.

Something like this works: 
```
<link rel="manifest" href="{{ url_for('static', filename='manifest.json') }}">

{
  "name": "OctoPrint",
  "icons": [
  {
    "src": "static/img/apple-touch-icon-144x144.png",
    "type": "image/png",
    "sizes": "144x144"
  }
  ],
  "start_url": "/",
  "background_color": "#000000",
  "display": "standalone",
  "orientation": "landscape"
}
```

~~The background colour is non-negotiable, any app that starts with a blinding white splash screen should be destroyed and the devs burned at the stake.~~

**Additional context**
Without the manifest, just a browser tab:
[image]

With the manifest, after using the "Add to Homescreen" in Chrome:
[image]

The splash-screen:
[image]

foosel, 2020-06-19:
Pushed to `maintenance` and ready for 1.4.1.

fieldOfView, 2020-06-19:
Why a black background though, when the default interface is bright white?

foosel, 2020-06-19:
At least the devices that I know lean towards dark stock UI schemes (notification bar and such). In that case a black background feels more "native". With so many people out there using Themeify with a dark theme it doesn't hurt in that regard either. And frankly, it's visible for such a short fraction of time that I don't feel like spending too much time on that decision ;)

fieldOfView, 2020-06-19:
ok

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel