OctoPrint/OctoPrint #3421: Your account is deactivated

mbelley, 2020-01-08:
I installed the latest rc version 1.4.0rc3
Add new user account. All is working fine until octoprint restart or reboot
once octoprint the user cannot login
the user get this: "Your account is deactivated".

Message in dev console: "POST http://hass-octo.local:5000/api/login 403 (FORBIDDEN)"
and message is "User unknown or password incorrect"

User was created in "users.yaml" and is still there.

foosel, 2020-01-13:
I just ran through the setup wizard with a fresh config, created a new user, restarted the server and everything works as expected. This is also one of my test scenarios that I run through prior to updating.

Can you reproduce this reliably? Can you **privately** share your `~/.octoprint/users.yaml` to gina@octoprint.org so I can take a look at it?

mbelley, 2020-01-16:
Work well with systemd

closed 2020-01-16 by mbelley