OctoPrint/OctoPrint #3423: typo in helptext (M108 vs M118)

JoGrob, 2020-01-10:
#### What were you doing?
1. "OctoPrint Settings" aufrufen
2. "Serielle Verbindung" aufrufen
3. "Firmware & Protokoll" aufrufen
4. Abschnitt "Notfallbefehle" betrachten

#### What did you expect to happen?
Die einbgetragenen Notfallbefehle (M112, M108, M410) sollten mit den darunter folgenden Erläuterungen zu Standardeinstellungen (M112, M118, M410) übereinstimmen.

#### What happened instead?
In den Erläuterungen zu Standardeinstellunge ist ein Tippfehler. Der Befehl M118 ist falsch. Richtig ist M108. Der Tippfehler ist sowohl in der deutsche als auch der englischen Spachdatei enthalten.

#### Did the same happen when running OctoPrint in safe mode?
n/a

#### Version of OctoPrint
1.3.12

#### Operating System running OctoPrint
n/a

#### Printer model & used firmware incl. version
n/a

#### Browser and version of browser, operating system running browser
n/a

#### Link to octoprint.log
n/a

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser
n/a

#### Screenshot(s)/video(s) showing the problem:
file: messages.po
line number: 5636
#: src/octoprint/templates/dialogs/settings/serialconnection.jinja2:375
msgid ""
"Use this to specify emergency commands which should jump any queues and -"
" if supported by the firmware - be force sent. Just the G or M code, "
"comma separated. Defaults to <code>M112</code>, <code>**_M118_**</code> and "
"<code>M410</code>."
msgstr ""
"Nutze dies um Notfallbefehle zu definieren, die interne Queues "
"überspringen und - falls von der Firmware unterstützt - forciert gesendet"
" werden sollen. Nur der G oder M Code, kommasepariert. Standard ist "
"<code>M112</code>, <code>**_M118_**</code> und <code>M410</code>."

####
I have read the FAQ.

LMS0815, 2020-01-10:
Screenshot(s)/video(s) showing the problem:
[image]
`M118` -> `M108 ` _Break and Continue_

foosel, 2020-01-13:
Duplicate of #3311 and fixed for 1.4.0. Thanks!

closed 2020-01-13 by foosel