OctoPrint/OctoPrint #3424: [Demande] Proposition de fonctionnalité intégrer à l'OS

Tam-Koowan, 2020-01-11:
Hello, I have been using Octoprint in wifi for two years now and I am delighted. I put a PC next to the printer very recently except that it does not have a wifi card so it cannot access the internet or the local network.
So I wanted to connect my PC to my RPI in RJ45 so that it could benefit from Internet connection except that it did not work.
The idea would therefore be to be able to integrate into the octoprint interface a function that would create a bridge between the wifi and the RPI network port, and tell it if you have internet from the wifi share it with the network.
Basically the idea would be to create an Ethernet router.
I hope I am quite clear because my English is not very good, it is translated from google.

thank you

Bonjour, j'utilise Octoprint en wifi depuis deux ans maintenant et j'en suis ravi. J'ai mis un PC à côté de l'imprimante très récement sauf qu'il n'a pas de carte wifi il ne peux donc pas accèder à internet ni au reseau local.
J'ai donc voulu connecter mon PC sur mon RPI en RJ45 pour qu'il puisse benificier de la connextion internet sauf que ça ne fonctionne pas.
L'idée serais donc de pouvoir intégrer dans l'interface d'octoprint une fonction qui permettrais de créé un pont entre le wifi et le port réseau du RPI, et de lui dire si tu as internet depuis le wifi le partager avec le reseau.
J'epsère que je suis assez claire car mon anglais est pas très bon, c'est traduit de google .

Merci à vous

BeatSlayer, 2020-01-12:
https://www.raspberrypi.org/forums/viewtopic.php?t=132674

Like this?

foosel, 2020-01-13:
That is something way too hardware and workflow specific to be integrated into core OctoPrint and rather suited for a plugin.

closed 2020-01-13 by foosel