OctoPrint/OctoPrint #3425: [1.4.0rc3] /api/settings accessible for guests

foosel, 2020-01-13:
@Lantoit thanks for testing. Attachments in emails don't come across in github issues, so I cannot see the images, but the text says enough. I think this is a sideeffect of the granular permission system, but lets see what @foosel has to say about this when she is back.

Edit: Confirmed with `curl http://octopi/api/settings` (ie: no API key or other headers specified, no client-side cache) that 1.4.0 RC3 returns the full settings dict and 1.3.12 returns a Forbidden error.

_Originally posted by @fieldOfView in https://github.com/foosel/OctoPrint/issues/3389#issuecomment-572678251_

foosel, 2020-01-21:
Now that was a bit more tricky than anticipated.

The "everything blocked for you" login dialog in 1.4.0* so far was done as part of the same page as the core UI. However, in order to initialize this in the first place, the settings are needed. After some brainstorming we are now back at the design of having said login dialog as a separate page like in 1.3.x with the forcelogin plugin, self contained and with a way less complicated initialization process.

That brings back the slightly annoying reload that's needed after login/logout but it makes everything way less of a horrible headache to maintain.

Patch ready for 1.4.0rc4

fieldOfView, 2020-01-21:
I checked out and am running 8d31d66, but `curl http://octopi.local/api/settings` - without any additional headers - still returns a full settings dict.

On a possibly related note, in 1.3.12 `curl http://octopi.local/api/login?passive=true` (again, without an API key) returns a 204 No Content status code, which I understand because there is no login. In 1.4RC3 as well as 8d31d66, it returns a dict:
`{"_is_external_client":false,"needs":{"group":["guests"]}}`

foosel, 2020-01-21:
Indeed... consider me slightly puzzled because I obviously tested this earlier and then it worked as expected. Investigating...

> In 1.4RC3 as well as [8d31d66](https://github.com/foosel/OctoPrint/commit/8d31d6687ec86539ee54e818cafa0463cfbe710c), it returns a dict:
> `{"_is_external_client":false,"needs":{"group":["guests"]}}`

That is intentional because in 1.4.0 even "no user" is an anonymous user which depending on the configuration might have some amount of access configured (if you change the permissions of the "read only" group).

foosel, 2020-01-21:
So apart from the wonky english in the corresponding commit message above this should now be fixed. Slightly embarrassing XD

foosel, 2020-01-21:
Also fixed a caching issue after restart that got introduced by the refactoring that took place as part of this fix.

fieldOfView, 2020-01-21:
It seems to work now with curl, and with a "fresh" browser. When the browser is logged in to OctoPrint, the X-Api-Key header is ignored and the browser uses the credentials of the session instead (which is fine).

foosel, 2020-01-28:
1.4.0rc4 is out

commit 3f3cb556 (2020-01-21): 🐛 Fix guest access to settings

Needed its own permission, which in turn needed some changes in how the
whole "guest without read permission" situation is handled. I've
reintroduced a custom login view since the main UI would not be able
to fully load up without settings access. Said login view will override
the default UI view if the account in question does not have
Permissions.STATUS and also Permissions.SETTINGS_READ. It pretty much is
the forcelogin plugin from 1.3.x.

Closes #3425
commit 95c3fa0b (2020-01-21): 🐛 Readd requires line that somehow got missing

Addendum to 3f3cb556a703f15c60907956e66f8b0abdef6a5d

Closes #3425
closed 2020-01-28 by foosel