OctoPrint/OctoPrint #3426: [1.40rc3] "Remember me" doesn't stick

foosel, 2020-01-13:
Not a huge issue, and probably part of the new permissions system, but I often find I've been logged out. I checked the "Remember me" button but it's seeming like my sessions only last about a day or so. I've clicked through the options to see if there is an auto-timeout configured anywhere and don't see anything like that. I have created a second user to test the access permissions, don't know if that might trigger an auto-logout, or if maybe the issue is possibly something on my browser. (Chrome running at latest version on mac)

If there is some sort of new auto-logout feature, I'd suggest having the ability to disable that, especially for single-user setups. I'm guessing many installs are a simple at-home setup (like mine) and most folks aren't going to want to login repeatedly.

_Originally posted by @ChrisHeerschap in https://github.com/foosel/OctoPrint/issues/3389#issuecomment-570921246_

ChrisHeerschap, 2020-01-13:
I've been seeing this consistently since reporting it - have been trying to rule anything out on my end. I run uBlock Origin and Privacy Badger on my systems and those _shouldn't_ be affecting things but I've disabled them on my 1.40rc3 page and can't say if it has consistently helped or had any effect.

Let me know anything I can test or try to contribute to fixing this. One thing that may be contributing is I usually hit my 1.40rc3 system from any of three laptops, one desktop system, and occasionally my phone. Don't know if the multiple logins might contribute. Even if it is something unique to my environment, it might be something you run into when 1.40 goes live and it would be good to know the workarounds.

ChrisHeerschap, 2020-01-15:
Well, it wasn't those two things - my 1.40 has been in an active tab all day long and I just switched to it and I was logged out.

CapnBry, 2020-01-17:
I've also been seeing this and have been halfheartedly investigating it too. When the `/api/login` POST happens, my remember parameter is always false regardless of the checkbox state in the UI. This goes back to `UiStateViewModel.login_remember` is always false too,

I don't know what the heck branch you're working on but [here's the problem right here in the staging/devel branch](https://github.com/foosel/OctoPrint/blob/staging/devel/src/octoprint/templates/index.jinja2#L68) right?
```
                 <input type="password" data-bind="value: login_password" class="input-block-level" placeholder="{{ _('Password')|edq }}">
                 <span class="pull-right"><small><a href="https://faq.octoprint.org/forgotten-password" id="login-forgotpassword" target="_blank" tabindex="-1">{{ _('Forgot password?') }}</a></small></span>
                 <label class="checkbox">
-                    <input type="checkbox" data-bind="value: login_remember"> {{ _('Remember me') }}
+                    <input type="checkbox" data-bind="checked: login_remember"> {{ _('Remember me') }}
                 </label>
                 <button class="btn btn-block btn-large btn-primary" type="submit" data-bind="click: login">{{ _('Log in') }}</button>
             </form>
```

foosel, 2020-01-17:
Well... and then @CapnBry comes along and just serves me the solution on a silver platter. I owe you a beer, maybe at MRRF? :) 

Fix is pushed and ready for 1.4.0rc4 👍

ChrisHeerschap, 2020-01-17:
@CapnBry's version of "half-heardedly investigating" appears to be _far more effective_ than mine. Thank you!

CapnBry, 2020-01-17:
I have to admit that when I started typing my response I didn't think I was going to be able to figure it out as I got lost in the hierarchy of javascript classes but Chrome's dev tools saved the day. That and Gina explaining the technicalities behind why it it would seem to work some of the time and all the time for some people. Team effort!

foosel, 2020-01-28:
1.4.0rc4 is out

commit 89b597a6 (2020-01-17): 🐛 Fix remember me checkbox

Closes #3426
closed 2020-01-28 by foosel