OctoPrint/OctoPrint #3427: uploaded gcode is changed (somehow) (Prusa MMU2, extruder, tool changes)

Mirarkitty, 2020-01-15:
#### What were you doing?
Sending gcode through octoprint (from PrusaSlicer)

1

#### Version of OctoPrint

Mid-2019 forward (??)

#### Issue

It seems like there's a difference on how the printer is handling tool switches if the print is run from Octoprint or if it's read from an SD card. The claim is that the extruder settings in Octoprint has an impact (I have yet to prove this myself).

Octoprint should NEVER change the gcode uploaded to the printer, regardless of it's own settings.

From around mid-2019 I've (and others) have had issues with tool switching gcode IF it has been sent through Octorprint. This could be due to octoprint upgrade or firmware upgrade on the printer side. (Impact is at the start of the print, where the tool switching on the MK3/MMU2 has really strange issues, including hanging and unloading filament without request.)

**CAN this change of gcode POSSIBLY happen in Octoprint**, and if so where would be a good place in the code to start looking for issues?

(The other possibility is that the MK3 is handling gcode from the serial port different from gcode from the sdcard.)

foosel, 2020-01-15:
Sounds very much like [this FAQ entry](https://community.octoprint.org/t/i-installed-octoprint-1-3-12-or-later-and-now-my-mmu2-doesnt-work-properly-the-whole-printer-resets-how-to-fix-this/12962), please check on your end that your printer profile is setup correctly. If it is, feel free to fill out the ticket template and request a reopen and we can look further into this.

> Octoprint should NEVER change the gcode uploaded to the printer, regardless of it's own settings.

Other people tell me OctoPrint should NEVER send gcode to the printer about which it knows it can cause damage and/or print ruining lock-ups. Tool change commands for non existing tools can cause both, depending on the firmware. I rather err on the side of caution since if something get damaged, it is usually me who gets to justify herself, regardless of where the fault lies.

closed 2020-01-15 by foosel

Mirarkitty, 2020-01-15:
This is what I've been reading up on as well, but those two seem to be in conflict. If there's gcode that might damage the printer, it would be best to refuse to send it at all and stop the print than to edit it. 

Which part of the code is/would be active in changing/blocking gcode? I'd like to look at it.

foosel, 2020-01-15:
> If there's gcode that might damage the printer, it would be best to refuse to send it at all and stop the print than to edit it.

And that's exactly what a misconfigured printer profile will do. It will refuse to send tool change commands (e.g. T1, T2, etc) to the printer for tool numbers it doesn't know actually exist, because on some printers that can cause crashes, hangs and so on.

Nothing is rewritten or modified, it simply doesn't send those commands, but that apparently causes all kind of weird issues like restarts or other odd stuff in the firmware side of things on the Prusa printers in particular, whereas on other multi extruder printers it simply causes the print to just come out in a single color. I have no idea why the Prusafork of Marlin is so sensitive to this, and frankly I also don't have to time to investigate that.

> Which part of the code is/would be active in changing/blocking gcode? I'd like to look at it.

https://github.com/foosel/OctoPrint/blob/8409fdaa48ff692bbb050086e8d7c09940b58c59/src/octoprint/util/comm.py#L3506-L3539

Mirarkitty, 2020-01-15:
Thanks. That looks like it SHOULD be safe enough. I'll see if I can investigate this on the Prusa side. It's REALLY annoying me... :p 

I think it would be a good idea to stop the print all together if a bad gcode is encountered (or make it an option not to care?) so you get a chance to configure it correctly.

Mirarkitty, 2020-06-06:
Something is definitely wrong. I have the correct setup (as in, it works every 4-5 tries), but it still blocks now and then. Can I ask for a feature to turn off this filter completely?

```
Send: N131 M403 E2 F0*38
Recv: Unknown G code: $128 G21
Recv: ok
Not queuing T3, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
Send: N132 M403 E3 F0*36
Recv: MMU <= 'F0 0'
Recv: ok
Recv: MMU <= 'F1 0'
```

[image]

Mirarkitty, 2020-06-06:
Ah, seems like changing temperature for tool 4 isn't great. Wonder what does that

```
Recv: ok
Send: N126 M105*34
Recv: echo:M104 Invalid extruder 4
T4 reported as invalid, reverting to T3
Recv: ok
Not queuing T3, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
Send: N127 G80*21
Recv: T:252.5 E:0 W:?
```

jneilliii, 2020-06-06:
@kantlivelong I believe knows about the MMU weirdness.