OctoPrint/OctoPrint #3428: Windows Py3 Unable to Install Plugins

jneilliii, 2020-01-16:
#### What were you doing?

Trying to install a plugin from the Plugin Manager in OctoPrint running in Python 3 under Windows. Additional discussion on the forum [here](https://community.octoprint.org/t/error-installing-plugin-could-not-parse-output-from-pip/14691).

#### What did you expect to happen?

The plugin to install.

#### What happened instead?

An error is reported "Could not parse output from pip, see plugin_pluginmanager_console.log for generated output".

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPrint

1.4.0rc3

#### Operating System running OctoPrint

Windows 10 Running Python 3.8.

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Chrome Version 79.0.3945.117 (Official Build) (64-bit)

#### Link to octoprint.log

[octoprint (3).log](https://github.com/foosel/OctoPrint/files/4073316/octoprint.3.log)
[plugin_pluginmanager_console (6).log](https://github.com/foosel/OctoPrint/files/4073317/plugin_pluginmanager_console.6.log)

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2020-01-20:
Thanks for the heads-up, should be fixed by the above commit, ready for 1.4.0rc4

foosel, 2020-02-11:
Totally forgot to close that on 1.4.0rc4 release...

commit 4460c8be (2020-01-20): 🐛 env needs to be native strings

So, bytes under py2 and unicode under py3.

Closes #3428
closed 2020-02-11 by foosel