OctoPrint/OctoPrint #3430: Request: Terminal textbox control scrolling and content filtering

lmcbmai, 2020-01-23:
With all 3 Suppress checkboxes selected and with Autoscroll option activated, even when nothing is being printed and no commands are entered manually, the content of terminal tab's textbox eventually moves up and disappears. This is very inconvenient during troubleshooting.

I'd suggest to keep Terminal tab's textbox unchanged, unless new unsuppressed line is added.
Additionally, would be nice to apply suppress filters only to newly added lines and keep existing text in the textbox unfiltered.

Thanks

hapklaar, 2021-04-12:
I second this request. I have a couple of filters in place, which also filter all movement lines. All the remaining lines I'm interested in disappear from the terminal window within seconds, making it unfit for keeping track of what actually happened. 

My terminal windows mostly looks like this during printing:

[image]

stefan-sherwood, 2022-08-01:
I have the same request. I was doing auto PID tuning, which takes some time, and if I wasn't looking at the screen at the moment the PID value lines came back they would soon be replaced with `[...]` even though there were no non-filtered lines following them.

stefan-sherwood, 2022-08-01:
I wrote a Python script that will give you a serial terminal to interact with the printer without seeing the noisy messages. 

```
from serial.tools import miniterm
import re
import sys

class OctoprintFilter( miniterm.Transform ):
        """Remove Octoprint messages"""
        def __init__( self ):
                self.filters = [ "Not SD printing", "wait", " T:\d*[.,]?\d* \/.*" ]

                self.rx_buffer = ""
                self.echo_buffer = ""

                self.echo_color = '\x1b[1m\x1b[31m' # bright red
                self.reset_color = '\x1b[0m'

                self.exclude = "|".join( [ x + "\n" for x in self.filters ] )

        def rx( self, text ):
                self.rx_buffer += text
                if text[ -1 ] != '\n':
                        return ""
                buffer, self.rx_buffer = self.rx_buffer, ""
                return re.sub( self.exclude, "", buffer )

        def echo( self, text ):
                if ord( text[-1] ) == 8:
                        self.echo_buffer = self.echo_buffer[:-1]
                else:
                        self.echo_buffer += text
                echo_buffer = self.echo_buffer
                if text[ -1 ] == '\n':
                        self.echo_buffer = ""
                return '\r' + self.echo_color + self.echo_buffer + '\x1b[0K' + self.reset_color

miniterm.TRANSFORMATIONS[ 'octoprint' ] = OctoprintFilter
sys.argv.extend( ['--echo', '--eol', 'CRLF', '--filter', 'octoprint'] )
miniterm.main( '/dev/ttyUSB0', 115200 )

```

- Copy this file to the device that is physically connected to the printer. In most cases, it will be a Raspberry Pi that you ssh into.
- Disconnect OctoPrint (via the Disconnect button on the OctoPrint web page) from the serial port.
- Run it with no arguments, e.g.
        `python printer_serial.py`