OctoPrint/OctoPrint #3431: Folder Test fails

willhoh, 2020-01-24:
#### What were you doing?
I test the folders under Settings > Folder

#### What did you expect to happen?
That the tests are completed positively

#### What happened instead?
When I test the folders under Settings > Folder, I get an error for each one that it is not writable. 

#### Did the same happen when running OctoPrint in safe mode?
Don't know, didn't done yet.

#### Version of OctoPrint
1.4.0rc3 but running with Python3

#### Operating System running OctoPrint
Rasbian On Pi B+ v1.2

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser
Chrome on Win10, Ubuntu 19.10 and Android

#### Link to octoprint.log
https://gist.github.com/willhoh/73be27baaaa57ccc782f194de1287396

#### Screenshot(s)/video(s) showing the problem:
[Settings > Folder](https://i.imgur.com/86R7Poa.png)

I have read the FAQ.

foosel, 2020-01-27:
Fixed by the above commit, ready for 1.4.0rc4

foosel, 2020-01-28:
1.4.0rc4 is out

commit 88ecfbdb (2020-01-27): 🐛 Fix another py2 vs 3 issue

Closes #3431
closed 2020-01-28 by foosel