OctoPrint/OctoPrint #3432: [RC Feedback] Feedback on the 1.4.0rc4 Release Candidate

foosel, 2020-01-28:
Please provide *general* feedback on your experience with the 1.4.0rc4 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

  * ❓ #3394 comm port autodetection still broken - appears to not be a regression and will get a closer look in 1.4.1

### Unrelated third party issues

### Issues needing analysis

schnello, 2020-01-28:
Hi.

Looks like the auto logon feature is brocken again.
[image]

nas@service-uplink:~$ cat .octoprint/config.yaml
accessControl:
    autologinAs: admin
    autologinLocal: true
    localNetworks:
    - 127.0.0.1/24
    - 192.168.1.0/24
    salt: TaRRqLPIYxxxxxxxxxxxxxxxxxxxxxx

foosel, 2020-01-28:
@schnello compared to what version? RC3? If so I have an idea what the issue might be.

schnello, 2020-01-28:
@foosel 
Yes RC3 was my last installed version

foosel, 2020-01-29:
@schnello Tracked as #3434

1n5aN1aC, 2020-01-30:
Just updated to RC4.

One hour-long print succeeded with no problems.  
Components tested:

- Upload gcode  
- Custom Actions defined in config  
- Custom Controls defined in config  
- Printing  
- GCode visualizer  
- Webcam  
- Various plugins

ciordia9, 2020-01-30:
USB Logitech C270 does not come back after Octopi reloads via forced or post-plugin update. It was happening in RC3 as well. Reboots are required after every update to get the stream operational again. (This may be endemic to the way USB works, I am not familiar) Hardware: B+

schnello, 2020-01-30:
Printed now some parts (12h) and works fine. no problems so far

foosel, 2020-01-30:
@ciordia9 Hm... The thing is, OctoPrint only really embeds the stream URL in the web interface, there's no way it can possibly affect it in any way. Either the stream works, or it doesn't, but OctoPrint cannot influence that in any way, on OctoPi it's (usually) mjpg-streamer that takes care of that and that is an entirely different process and not restarted or anything when OctoPrint is due to whatever reason.

What exactly constitutes "does not come back"? The view stays black? Is the LED on the camera still on? Is it powered through the Pi or from a powered USB hub? Is your Pi maybe running with active undervoltage alerts? What I could imagine happening is that the rather resource intensive task of running an update causes a voltage drop that makes the webcam disconnect from the USB. Would be interesting to know if simply unplugging and replugging it would also make it work again (after  ~2min when webcamd will try to fire it up again). In any case, this is most likely an issue either with your setup, or with OctoPi, but not anything I see how I could be responsibe for and/or fix from OctoPrint given that I can only embed an URL in a web interface and that's it.

ciordia9, 2020-01-30:
> What exactly constitutes "does not come back"? The view stays black? Is the LED on the camera still on? Is it powered through the Pi or from a powered USB hub? Is your Pi maybe running with active undervoltage alerts? 

I'll validate. I don't recall ever having to do this in the stable branch is why I brought it up here. I'll gather more information.

gege2b, 2020-01-30:
I'm testing right now, and no problem as far as I can tell

MSeal, 2020-02-06:
Did the RC files get removed from PyPI? `pip` can't find the rc4 release version and curling curling `https://pypi.org/pypi/OctoPrint/json` showed `1.4.0rc4` in the `"releases"` section was `[]` instead of including the release content.

foosel, 2020-02-06:
I made a mistake with the release builds for pypi (I really need to automate this stuff) and since pypi doesn't allow to replace uploaded files without a version increment I decided to just delete the broken stuff again and not jump through the hoops of post1 version numbering for an RC.

CapnBry, 2020-02-06:
For posterity: "Remember me" login functionality was also broken by the forcelogin view introduced in rc4. If you closed the browser window, the remember cookie wouldn't be enough to get you back in next time. It is also fixed by a88d95af77b721cac52f15092bd225773687dd79 from #3434 so this will be resolved by rc5.

Farami, 2020-02-06:
I'm in the process of doing the first print with rc4 and the gcode viewer is lagging behind quite a lot.

foosel, 2020-02-07:
@Farami didn't touch the GCODE viewer at all. But how much in sync it is with the actual print can definitely rely on buffer sizes in the firmware & network speed.

Just took a look at a print job currently running on one of my printers under rc4 and it looks as expected here 🤔

ChrisHeerschap, 2020-02-07:
Installed rc4 when it was available. 

Running from octopi 0.17.0 install on:
* Pi 4 1GB
* Pi 3 
* Pi B+ 

That original Pi B+ is honestly far too slow to be usable as an octoprint system - pretty minimal install (almost no add-on plugins) and the raspi native camera running via octopi-supplied mjpg-streamer and the system is slow enough to visibly affect print quality. The other two are obviously far more than strong enough.

Have ran off many prints and so far no complaints or observed issues. Not getting the repeated logouts and forgotten "Remember me" mentioned in rc3, so that's good.

One minor issue noted that may or may not be anything - had a printer error last night (failed print, filament in hot end fan, Prusa errored out) and after rectifying that and reconnecting to the printer, I reloaded the Octoprint page and my settings for sorting the gcode files and only showing files stored locally had both been cleared. Trivial to reset so not a big deal. Can try to reproduce if further investigation is merited.

ChrisHeerschap, 2020-02-11:
Spoke too soon - I've been accessing one of my 1.40rc4 systems from several macs and an iPhone and have been experiencing random logouts without anything being remembered - and I've always been clicking "Remember me". Seems like when I get enough clients connected (not concurrently, I move from system to system and am a single user hitting the octoprint instance) some of the logins get expired or invalidated. Can create a ticket if it seems useful, and provide logs.

CapnBry, 2020-02-11:
Try 1.4.0rc5 @ChrisHeerschap, there was a issue in rc4 that broke a different part of the login process so it would only remember you as long as the browser window was open (or cached in the background in the case of a mobile device).

ChrisHeerschap, 2020-02-11:
Yup, just saw the notification of the release so am updating all my systems now.

foosel, 2020-02-11:
Y'all also just reminded me that I forgot to close this ticket, new one is #3450 ^^

closed 2020-02-11 by foosel