OctoPrint/OctoPrint #3433: [Request] Allow sorting of printables by last printed date & add "favourite" flag

foosel, 2020-01-28:
Feature request from a Jesse on Patreon:

> [I]t would be helpful if I could sort by Last Printed date [...]. Is there a way to do this? [...] I print the same models repeatedly [...]. I have sliced these models soooo many times with different settings but there are some gcode files that are perfect for the printers I use them on. I always end up printing the same ones. It would be amazing to have the ability to bookmark files in the browser so that I can sort by that mark instead (kind of like a favorite/hear button) so that I could easily distinguish the preferred files from the ones that aren't so good.

tldr:

  * new file sorting option "by last printed date"
  * ability to favourite files
  * new file filtering option "only favourited"

metalfacesec, 2020-02-04:
@foosel Would you mind if I take a shot at this one? This would be my first contribution so I may need a little time to get it fully done to your standards but, I was messing around last night and added the ability to add metadata to files for a 'favorite' flag. Seems to work pretty good from my initial testing.

foosel, 2020-02-04:
@metalfacesec sure, go ahead!

metalfacesec, 2020-02-04:
Awesome, thank you @foosel ! I'll let you know if I have any questions along the way!

foosel, 2020-06-08:
@metalfacesec Are you still on this?

foosel, 2020-06-18:
Apparently not :/

foosel, 2020-06-22:
Sorting by last print date is now implemented on `maintenance`. Favourite function should be implemented via a plugin.