OctoPrint/OctoPrint #3434: [1.4.0rc4] Autlogin via IP address is broken due to return to forcelogin view

foosel, 2020-01-29:
Looks like the auto logon feature is broken again.
[image]
Same wih IP:
[image]

```
nas@service-uplink:~$ cat .octoprint/config.yaml
accessControl:
    autologinAs: admin
    autologinLocal: true
    localNetworks:
    - 127.0.0.1/24
    - 192.168.1.0/24
    salt: TaRRqLPIYxxxxxxxxxxxxxxxxxxxxxx
```

_Originally posted by @schnello in https://github.com/foosel/OctoPrint/issues/3432#issuecomment-579340581_

LMS0815, 2020-01-29:
Can confirm issue appearing although user-management is disabled.
OctoPrint keeps asking for log - same computer different browser (same IP address).

foosel, 2020-01-29:
Autologin and disabled access control situation should be fixed by the above commit, ready for 1.4.0rc5

schnello, 2020-01-29:
@foosel 
Thx for the quick fix.
I merge the fix at my local rep manually...
the good news:
Autologin works again and no password is required in my local network.

the bad new:
I needs a lot of time. A look into the log files shows some errors regarding "octoprint.util.net"
`2020-01-29 15:45:20,268 - octoprint.server.util.flask - ERROR - Could not autologin user admin from None for networks IPSet(['127.0.0.0/24', '192.168.1.0/24', '::ffff:127.0.0.0/120', '::ffff:192.168.1.0/120'])
Traceback (most recent call last):
  File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 594, in determine_user
    if netaddr.IPAddress(remote_address) in local_networks:
  File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/netaddr/ip/__init__.py", line 306, in __init__
    'address from %r' % addr)
AddrFormatError: failed to detect a valid IP address from None
2020-01-29 15:45:20,272 - octoprint.util.net - ERROR - Error while trying to determine whether None is a local address
Traceback (most recent call last):
  File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/net.py", line 49, in is_lan_address
    address = unmap_v4_as_v6(address)
  File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/net.py", line 112, in unmap_v4_as_v6
    if address.lower().startswith("::ffff:") and "." in address:
AttributeError: 'NoneType' object has no attribute 'lower'
`

Note: This access was via an SSH Tunnel. I think that shouldn't make a difference. But will test it again at home.

foosel, 2020-01-29:
I take it you only merged the one commit referenced here? If so, also merge 010221ab5b3b8e2e994897a9811e7bba4333c958

schnello, 2020-01-29:
Oh my fail. Yes with 010221a all is fine

CRCinAU, 2020-01-30:
Can confirm login works again with user access control disabled after applying the following patches:

https://github.com/foosel/OctoPrint/commit/a88d95af77b721cac52f15092bd225773687dd79.patch
https://github.com/foosel/OctoPrint/commit/010221ab5b3b8e2e994897a9811e7bba4333c958.patch

Just had to modify the file path in the files to match the installed version.

foosel, 2020-02-11:
1.4.0rc5 is released.

commit a88d95af (2020-01-29): 🐛 Fix autologin

We need to actually try to identify the user just like we used to do
with the forcelogin plugin. Just checking permissions will only work
if we already have a session.

Closes #3434
closed 2020-02-11 by foosel