OctoPrint/OctoPrint #3435: [1.4.0rc5.dev1+g6471a04a] Browser fails to load

gdombiak, 2020-01-29:
#### What were you doing?

1. Tried to load home page
4. See error

[image]

See JavaScript errors below.

#### What did you expect to happen?

Load the page :)

#### What happened instead?

Keeps saying connecting

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.0rc5.dev1+g6471a04a

branch: staging/deve
last commit: 6471a04a869299f2cdd803f1fa9a9006843cc4c6

#### Operating System running OctoPrint

MacOS 10.14.6

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Chrome 79.0.3945.130 (latest)

Safari  13.0.4

#### Link to octoprint.log

https://gist.github.com/gdombiak/29caa28fde23f9d78b64a8a8e48c5876

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

> [DOM] Found 2 elements with non-unique id #settings-serverOnlineCheckHost: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-small" data-bind=​"value:​ server_onlineCheck_host" id=​"settings-serverOnlineCheckHost">​ <input type=​"text" class=​"input-small" data-bind=​"value:​ server_onlineCheck_host" id=​"settings-serverOnlineCheckHost">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-serverOnlineCheckInterval: (More info: https://goo.gl/9p2vKq) <input type=​"number" min=​"1" step=​"1" class=​"input-mini" data-bind=​"value:​ server_onlineCheck_interval" id=​"settings-serverOnlineCheckInterval">​ <input type=​"number" min=​"1" step=​"1" class=​"input-mini" data-bind=​"value:​ server_onlineCheck_interval" id=​"settings-serverOnlineCheckInterval">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-serverOnlineCheckPort: (More info: https://goo.gl/9p2vKq) <input type=​"number" min=​"1" max=​"65535" step=​"1" class=​"input-small" data-bind=​"value:​ server_onlineCheck_port" id=​"settings-serverOnlineCheckPort">​ <input type=​"number" min=​"1" max=​"65535" step=​"1" class=​"input-small" data-bind=​"value:​ server_onlineCheck_port" id=​"settings-serverOnlineCheckPort">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-serverRestartCommand: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-block-level" data-bind=​"value:​ server_commands_serverRestartCommand" id=​"settings-serverRestartCommand">​ <input type=​"text" class=​"input-block-level" data-bind=​"value:​ server_commands_serverRestartCommand" id=​"settings-serverRestartCommand">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-systemRestartCommand: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-block-level" data-bind=​"value:​ server_commands_systemRestartCommand" id=​"settings-systemRestartCommand">​ <input type=​"text" class=​"input-block-level" data-bind=​"value:​ server_commands_systemRestartCommand" id=​"settings-systemRestartCommand">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-systemShutdownCommand: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-block-level" data-bind=​"value:​ server_commands_systemShutdownCommand" id=​"settings-systemShutdownCommand">​ <input type=​"text" class=​"input-block-level" data-bind=​"value:​ server_commands_systemShutdownCommand" id=​"settings-systemShutdownCommand">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-webcamFfmpegPath: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-block-level" data-bind=​"value:​ webcam_ffmpegPath, valueUpdate:​ 'afterkeydown'" id=​"settings-webcamFfmpegPath">​ <input type=​"text" class=​"input-block-level" data-bind=​"value:​ webcam_ffmpegPath, valueUpdate:​ 'afterkeydown'" id=​"settings-webcamFfmpegPath">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-webcamSnapshotUrl: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-block-level" data-bind=​"value:​ webcam_snapshotUrl, valueUpdate:​ 'afterkeydown'" id=​"settings-webcamSnapshotUrl">​ <input type=​"text" class=​"input-block-level" data-bind=​"value:​ webcam_snapshotUrl, valueUpdate:​ 'afterkeydown'" id=​"settings-webcamSnapshotUrl">​
(index):1 [DOM] Found 2 elements with non-unique id #settings-webcamStreamUrl: (More info: https://goo.gl/9p2vKq) <input type=​"text" class=​"input-block-level" data-bind=​"value:​ webcam_streamUrl, valueUpdate:​ 'afterkeydown'" id=​"settings-webcamStreamUrl">​ <input type=​"text" class=​"input-block-level" data-bind=​"value:​ webcam_streamUrl, valueUpdate:​ 'afterkeydown'" id=​"settings-webcamStreamUrl">​
packed_libs.js?3339d949:3819 

> jQuery.Deferred exception: Cannot read property 'mobile' of undefined TypeError: Cannot read property 'mobile' of undefined
    at http://localhost:5000/static/webassets/packed_core.js?c87bb82a:16976:48
    at HTMLDocument.<anonymous> (http://localhost:5000/static/webassets/packed_core.js?c87bb82a:17035:11)
    at mightThrow (http://localhost:5000/static/webassets/packed_libs.js?3339d949:3535:29)
    at process (http://localhost:5000/static/webassets/packed_libs.js?3339d949:3603:12) undefined

> jQuery.Deferred.exceptionHook @ packed_libs.js?3339d949:3819
process @ packed_libs.js?3339d949:3607
setTimeout (async)
(anonymous) @ packed_libs.js?3339d949:3641
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
fire @ packed_libs.js?3339d949:3407
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
ready @ packed_libs.js?3339d949:3879
completed @ packed_libs.js?3339d949:3889
packed_libs.js?3339d949:3828 Uncaught TypeError: Cannot read property 'mobile' of undefined
    at packed_core.js?c87bb82a:16976
    at HTMLDocument.<anonymous> (packed_core.js?c87bb82a:17035)
    at mightThrow (packed_libs.js?3339d949:3535)
    at process (packed_libs.js?3339d949:3603)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

gdombiak, 2020-01-29:
I removed venv and Library/ApplicationSupport/OctoPrint to ensure a clean install.

gdombiak, 2020-01-29:
$.browser  --> returns undefined so $.browser.mobile then fails

foosel, 2020-01-30:
Hm... I cannot reproduce this here... Exact same version loads just fine (and is also pretty much what was released as 1.4.0rc4 and based on the tracking that doesn't look like it's unusable either), tested Chrome (identical version) and Firefox.

And:

[image]

Can you please tell me what you have in `config.yaml` for

  * `devel.stylesheet`
  * `devel.cache.enabled`
  * `devel.webassets.bundle`
  * `devel.webassets.clean_on_startup`

---

As a sidenote, since the 1.4.0 RCs have now moved over to the maintenance RCs release channel, the continued work on fixes and thus for those has also moved to `staging/maintenance`. I try to keep `staging/devel` in sync, but it's not a top priority tbh. If you want to live on the bleeding edge, switch to `staging/maintenance` now.

gdombiak, 2020-01-31:
No idea what was going on with this computer. As you, my other mac has no issue. Got tired so deleted everything and started from scratch. Things are fine now. Who knows what was old/corrupted.

Thanks Gina for your help & time.
Gaston

closed 2020-01-31 by gdombiak