OctoPrint/OctoPrint #3437: [bug] Octoprint stuck on 'saving' when uploading a file to print.

Tob3ster97, 2020-01-30:
#### What were you doing?

Uploading file to OctoPrint.

#### What did you expect to happen?
File to  be uploaded for printing.

#### What happened instead?
'Saving' stuck. Stays at Saving forever.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
OctoPrint 1.3.12 

#### Operating System running OctoPrint
OctoPi (0.17.0)

#### Printer model & used firmware incl. version
CR10s Pro. Stock Firmware

#### Browser and version of browser, operating system running browser
Chrome (Version 79.0.3945.130 (Official Build) (64-bit))

#### Link to octoprint.log
https://gist.github.com/Tob3ster97/9fc597102674e182477a956915367449

#### Link to contents of terminal tab or serial.log
n/a serial logging not enabled.

#### Link to contents of Javascript console in the browser
https://gist.github.com/Tob3ster97/9fc597102674e182477a956915367449

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Tob3ster97, 2020-01-30:
[image]

Steeven9, 2020-02-03:
Same problem here. The file will eventually be uploaded but after ~5 minutes stuck on `Saving...`
OctoPrint 1.4.0rc4 on win7 32bit, Ender 3 with Marlin 2.0.1

foosel, 2020-02-03:
Can both of you please share a file with which you can reproduce the behaviour? 

And can you @Steeven9 also share octoprint.log and the contents of your browser's error console as well as a screenshot of the network tab focused on the upload request?

Tob3ster97, 2020-02-03:
[Mini_caballete_easel_no_support.zip](https://github.com/foosel/OctoPrint/files/4150426/Mini_caballete_easel_no_support.zip)

Tob3ster97, 2020-02-03:
The attached file isnt the original one, howevever the same issue is seen. Stuck on saving and sometimes with eventually upload after 5+ minutes. Some files can take over 30 minutes when i leave it on. And i do have fast internet so i assume this isnt the issue.

Steeven9, 2020-02-03:
As for the file causing the issue, I tried with 10 different models and it happened with all of them, here's one: [Phone_Stand03.txt](https://github.com/foosel/OctoPrint/files/4150379/Phone_Stand03.txt) (converted to txt since GitHub doesn't accept .gcode)

[octoprint.log](https://github.com/foosel/OctoPrint/files/4150415/octoprint.log) (canceled out IP addresses)

Browser console tab (Chrome Version 79.0.3945.130 (Official Build) (64-bit)):
[image]

Network tab while waiting:
[image]

Network tab when done:
[image]

Tob3ster97, 2020-02-03:
Also OctoPrint as a whole is extemely slow. can take 5-30 minutes to load the web interface and the same again to login.

Steeven9, 2020-02-04:
> Also OctoPrint as a whole is extemely slow. can take 5-30 minutes to load the web interface and the same again to login.

Maybe it helps, maybe it doesn't, but since I updated to rc3 (and then rc4), the OctoPrint webpage on load shows "Loading OctoPrint's UI, please wait...", while it never did before on stable versions. 
Could it be somehow realted to this?

Also perhaps it's worth noting that I had no issues whatsoever with Python 2.7 and the latest stable version. Then I updated to Python 3 and rc3 and this popped up

xthedakmanx, 2020-04-05:
+1 having same issue still

StevoP80, 2020-04-26:
+1 I have the same issue. Seems to happen randomly. I run multiple instances of octoprint on a single raspberry pi, and can upload the same file to multiple instances and it may hang on one instance but work on another. Does not seem to be file dependent.

johncmolyneux, 2020-05-15:
This has started happening for me today, just out of the blue - no plugins have changed (I have very few of them), no system changes to the Pi, and nothing different than yesterday when it worked fine.

fieldOfView, 2020-05-15:
To everybody who has "the same issue", we need logs or we can not see what happened.

johncmolyneux, 2020-05-15:
Of course.  What logs and how do I get them?

fieldOfView, 2020-05-15:
As requested by @foosel above:
> octoprint.log and the contents of your browser's error console as well as a screenshot of the network tab focused on the upload request

See this FAQ entry about where to find the logs:
https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

foosel, 2020-06-12:
The only thing I've been able to reproduce so far that looked familiar turned out to be caused by a test plugin I had written for something else. So... cannot reproduce, need more logs and test samples.

bwilliams18, 2020-06-16:
I'm encountering the same issues. Below are the log outputs. Looks like there's a 500 error from the Tornado api calls.

`2020-06-16 03:03:50,805 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|      cores: 4
|      freq: 1500.0
|      ram: 4041732096
|  os:
|      id: linux
|      platform: linux2
|  plugins:
|      pi_support:
|          model: Raspberry Pi 4 Model B Rev 1.2
|          octopi_version: 0.17.0
|  python:
|      pip: 19.3.1
|      version: 2.7.16
|      virtualenv: /home/pi/oprint
2020-06-16 03:03:50,811 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2020-06-16 03:03:50,824 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2020-06-16 03:03:51,236 - octoprint.server - INFO - Shutting down intermediary server...
2020-06-16 03:03:51,626 - octoprint.server - INFO - Intermediary server shut down
2020-06-16 03:03:51,628 - octoprint.events - INFO - Processing startup event, this is our first event
2020-06-16 03:03:51,629 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2020-06-16 03:03:51,631 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2020-06-16 03:03:51,631 - octoprint.server - INFO - Autoconnect on startup is configured, trying to connect to the printer...
2020-06-16 03:03:51,669 - octoprint.filemanager.analysis - INFO - Starting analysis of local:1001.gcode
2020-06-16 03:03:51,671 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/1001.gcode
2020-06-16 03:03:51,702 - octoprint.filemanager - INFO - Added 2 items from storage type "local" to analysis queue
2020-06-16 03:03:51,712 - octoprint.server - INFO - Trying to connect to configured serial port AUTO
2020-06-16 03:03:51,743 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-06-16 03:03:51,748 - octoprint.server.util.watchdog - INFO - Running initial scan on watched folder...
2020-06-16 03:03:51,750 - octoprint.server.util.watchdog - INFO - ... initial scan done.
2020-06-16 03:03:51,760 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi' for _http._tcp
2020-06-16 03:03:51,770 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi' for _octoprint._tcp
2020-06-16 03:03:51,811 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2020-06-16 03:03:51,894 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2020-06-16 03:03:51,897 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Detecting baudrate"
2020-06-16 03:03:51,919 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2020-06-16 03:03:52,691 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-06-16 03:03:52,697 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-06-16 03:03:52,735 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-06-16 03:03:52,912 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-06-16 03:03:53,039 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2020-06-16 03:03:53,498 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-06-16 03:03:55,170 - octoprint.filemanager.analysis - INFO - Analysis of entry local:1001.gcode finished, needed 3.50s
2020-06-16 03:03:55,882 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/ (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-06-16 03:03:55,883 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-06-16 03:03:55,892 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2020-06-16 03:03:55,961 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.1.0.129/'}
2020-06-16 03:03:56,175 - octoprint.filemanager.analysis - INFO - Starting analysis of local:desk-bracket.gcode
2020-06-16 03:03:56,177 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /home/pi/.octoprint/uploads/desk-bracket.gco$
2020-06-16 03:03:59,405 - octoprint.filemanager.analysis - INFO - Analysis of entry local:desk-bracket.gcode finished, needed 3.23s
2020-06-16 03:04:01,105 - octoprint.server.preemptive_cache - INFO - ... done in 5.14s
2020-06-16 03:04:01,105 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://octopi.local/'}
2020-06-16 03:04:02,007 - octoprint.server.preemptive_cache - INFO - ... done in 0.90s
2020-06-16 03:04:02,322 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:10.1.2.172
[... cut ...]
2020-06-16 03:12:28,482 - tornado.access - ERROR - 500 GET /api/files?recursive=true (::ffff:10.1.2.172) 23.52ms
2020-06-16 03:12:28,688 - tornado.access - ERROR - 500 GET /api/files?recursive=true (::ffff:10.1.2.172) 10.70ms
2020-06-16 03:14:48,363 - tornado.access - ERROR - 500 GET /api/files?recursive=true (::ffff:10.1.2.172) 9.60ms
2020-06-16 03:15:08,828 - tornado.access - ERROR - 500 GET /api/files?recursive=true (::ffff:10.1.2.172) 13.84ms
2020-06-16 03:15:09,037 - tornado.access - ERROR - 500 GET /api/files?recursive=true (::ffff:10.1.2.172) 10.56ms`

EddieGreen, 2020-06-22:
I've experienced the same issue for a long time.

The cause for this on my install was the plugin [DisplayLayerProgress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/). 

This plugin seems to take quite a long time to process anything but the smallest file. Until it has finished, upload progress will remain 'saving...'

Can anyone else confirm that they have this plugin installed and, by disabling it, speed of upload is restored to normal?

foosel, 2020-06-22:
I recently pushed a performance fix for the wrapper that plugins like that one need to use for processing files on upload, so maybe that issue is now actually fixed as far as possible.

EddieGreen, 2020-06-22:
Do you think that this adresses my concerns and potential fixes suggested over at [DisplayLayerProgress](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/154#issue-642895896)?

foosel, 2020-06-22:
It adds an optimization, however it still doesn't solve really bloody expensive processing inside a third party plugin, it can only optimize the stuff around that. DLP still probably should have that investigated.

I found that by switching one of my own plugins wrapping the file uploads from using regex matching to substring search and modifying the logic slightly I could significantly improve on performance.

EddieGreen, 2020-06-22:
Thanks for this explanation.

For this and another issue - stalling prints during upload - it looks like OctoPrint gets the blame for processor intensive tasks within plugins and, like me, folks run around looking for bugs and fixes to OctoPrint.

Does OctoPrint's plugin interface require developers to allow its current tasks to complete and report before they take over? Or to require them to report the plugin's status such as starting, processing, and completion? 

I don't see a status bar in the main interface but it could be implemented and made a requirement. Sure would reduce all the head-scratching in such cases as this.

foosel, 2020-06-22:
> Does OctoPrint's plugin interface require developers to allow its current tasks to complete and report before they take over? Or to require them to report the plugin's status such as starting, processing, and completion?

This interface doesn't. And adding that would also be non trivial. 

In general, OctoPrint does have a safe mode precisely so that users can disable all third party plugins with one central switch and check if a problem persists in that mode. If it doesn't -> third party plugin issue, not core OctoPrint.

Sadly way too many people ignore this request in the bug template and also the debugging steps outlined on the forum.

EddieGreen, 2020-06-22:
> OctoPrint does have a safe mode...

Good point.

iplayfast, 2020-07-05:
I recently had the same problem. Which happened after completing a print and starting the next.
 I noticed that swap was being used so turned that off. (sudo swapoff -a)
That made little difference. The big problem seemed to be wifi connection. Pinging back to my computer (I was connected through ssh) took 450ms. 
Giving up, I decide to reboot. (sudo shutdown -r now) 
Which seemed to clear whatever error there was. 
Back to normal now.

foosel, 2020-07-22:
Still unreproduced on my end, however, as mentioned 1.4.1 contains improvements in the upload pipeline that might solve this. Please test against the 1.4.1 RCs that are out already, if I don't hear anything about this still being an issue against the RCs I'll close this on release of 1.4.1.

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel