OctoPrint/OctoPrint #3438: [Request] On login, delay newsfeed refresh (by 60 seconds+)

OutsourcedGuru, 2020-01-31:
**Is your feature request related to a problem? Please describe.**
I log into my interface on a daily basis. On days like today when there are positive hits on the newsfeed(s), the interface becomes unresponsive until the popup has been generated and dismissed(?)

- **Connection** side panel widget and related auto-connect behavior doesn't happen until this has been accomplished
- **Terminal** registers no serial detection and Hello-related activity
- **Files** side panel widget in the past has been unresponsive during this activity as I recall

During this newsfeed-related activity I am unsure of whether the connection has occurred or not. It's quite possible that the connection _did_ occur early on but none of this is bubbling up to the web interface yet.

There are many times when this happens and I think that I now need to start troubleshooting things. The popup then happens, I dismiss it and then suddenly I realize that there is no actual problem.

**Describe the solution you'd like**
I would like the web interface to become fully responsive as soon as it can. I would like attached client services to quickly know that the printer board has been connected.

I would like the updater and newsfeed mechanisms to "take a back seat" to the other activities going on in the interface.

I am guessing that soon there will be a wave of new plugin submissions and versions to accompany the Py3-related requirements. Although updates are important, it would be nice if they wouldn't step on the use of the interface upon startup.

I'm thinking that _if_ the newsfeed check occurs a full minute _after_ a ready-to-go web interface is witnessed by the user then this would be better. Although OctoPrint releases—to me—are important, the accumulation of plugin-related updates and introductions happen too frequently to get such high priority, in my humble opinion.

**Describe alternatives you've considered**
I've considered just turning off all the newsfeeds in the Settings.

**Additional context**
N/A

foosel, 2020-02-04:
The problem with delaying these popups is that then they might pop up just when the user least expects it and cause annoying "misclicks". Since the announcements are also used for important messages like "security issue, update ASAP" kinda stuff, a delay could pose an additional problem (user finishes what they need to be done before the display timeout runs out).

I'm wondering about the actual reason for the issues you are seeing here though, because I honestly cannot say that I've ever observed noticeable delays on my own machines (even though I get a significant number of notifications on these too), and definitely not when it comes to state reporting to the UI, which usually should go through the web socket.

I'm wondering if you might be running on a fallback transport for the push socket that's limited by the 2-concurrent-requests-per-host-limitation imposed by the browser and thus fighting with things like "fetch announcements" or "check if there are updates".

The next time this happens, could you do me the favour to open your network tab and check if OctoPrint is doing xhr polling against the sockjs endpoint or using the websocket? Should also generate an entry in the error console if it falls back to a different transport.

OutsourcedGuru, 2020-02-26:
Each "first visit" of the day to the web interface recently results in a really long delay until the interface is fully functional. (I haven't applied the 1.4.0rc6 yet in this case.) So it's really grinding on that newsfeed.

In less-frequent cases, the newsfeed lookup is just failing. My own Internet connectivity here is pretty good (205Mbps) so it's not _that_. I don't know if this is load on your newsfeed server but that's a possibility.

foosel, 2020-02-26:
Still haven't seen this happening myself even remotely. Info on whether it's on xhr polling or websocket would help.

jneilliii, 2020-02-26:
@OutsourcedGuru, does it happen if your OctoPrint is running in safe mode?  I've noticed the same delay but attributed it to my tplink plugin because it checks the statuses (albeit theoretically asynchronously) of all the configured plugs on connection.

foosel, 2020-02-28:
*(Wanted to post this yesterday already, but Github was throwing nothing but errors at me)*

Just checked the loading times on a 1.4.0rc6 install in safe mode. Full page without caching loads within 10s, with caching it's at 3-4s.

The request to the endpoint of the announcements plugin (which is what's being blamed here) takes 1.4s and based on the waterfall diagram doesn't appear to block further communication.

I'd honestly prefer to treat this as a bug report and ask you to provide the usual set of logs and environment information, @OutsourcedGuru. The behaviour you *seem* to be seeing strikes me as very much unusual.

There are also no connectivity or uptime issues with the server in question according to my monitoring (see also https://status.octoprint.org).

fieldOfView, 2020-02-28:
> (see also https://status.octoprint.org).

Ofcourse that's a thing... I am impressed as always by how organised your whole setup is, though I am appalled by the lack of radial gradient on that page ;-)

OutsourcedGuru, 2020-03-20:
@jneilliii Seems to, yes.

I'll re-spin this as a bug if you're sure you want that.

**Any guidance on resetting anything within config.yaml so that I know it will poll the server on the next boot?** I can then force it, having setup my browser's Developer's Console so that I can profile the downloads.

jneilliii, 2020-03-20:
I think it's probably somewhere here...
```
plugins:
    announcements:
        _config_version: 1
        channels:
            _blog:
                read_until: 1584628200
            _important:
                read_until: 1521111600
            _octopi:
                read_until: 1573722900
            _plugins:
                read_until: 1584489600
            _releases:
                read_until: 1583322300
```