OctoPrint/OctoPrint #3439: Approx. Total Print Time is empty while printing via API

ochm, 2020-02-01:
When pritn starts directly from API, the Approx. Total Print Time on left panel is "-".
When the print is loaded to Octoprint and started manualy from Octoprint, everything looks good.

#### What did you expect to happen?
Display Approx. Total Print Time in both situations.

#### What happened instead?
Approx. Total Print Time is not dissplayed 

#### Did the same happen when running OctoPrint in safe mode?
Never tried.

#### Version of OctoPrint
OctoPrint 1.3.12

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
MK3S 3.9.0-beta-3175

#### Browser and version of browser, operating system running browser
Chrome, latest, W10

Thank you.

ochm, 2020-02-01:
I really believe that submitted informations are enough to test and repeat the issue.

fieldOfView, 2020-02-01:
Your report is missing logs.

How are you using the API? Through eg Cura and the OctoPrint Connection plugin or through your own code? If the latter, is that code available somewhere?

foosel, 2020-02-03:
Approximate print time can only be displayed if OctoPrint has had a chance to actually analyse the uploaded file. If you start the print right during our after upload, that won't be the case. 

The documented API is the exact same that OctoPrint's own web interface uses.

But as @fieldOfView said, I need more information here for analysis.

ochm, 2020-02-03:
OK, what further informations do you need? Some gcode, or some logs?

ochm, 2020-02-03:
Gcode is sent directly from PrusaSlicer-2.1.1, the logs i'll upload later, gcode doesn't mater, the behavior is still the same, but here is some sample.
[3DBenchy_0.1mm_CPE_MK3S_3h10m.gcode.zip](https://github.com/foosel/OctoPrint/files/4149806/3DBenchy_0.1mm_CPE_MK3S_3h10m.gcode.zip)

and here are logs
[logs.tar.gz](https://github.com/foosel/OctoPrint/files/4149802/logs.tar.gz)

Gcode sent from Slicer to Octoprint (without Start print after upload), Octoprint displays after short time:
File: 3DBenchy_0.1mm_CPE_MK3S_3h10m.gcode 
Uploaded: 2020-02-03 20:25:30
Timelapse: -
Filament (Tool 0): 4.28m / 10.29cm³
Approx. Total Print Time: 2.5 hours

But with Start print after upload i see only:
File: 3DBenchy_0.1mm_CPE_MK3S_3h10m.gcode 
Uploaded: 2020-02-03 20:28:06
User: API client
Timelapse: -
Approx. Total Print Time: -

BUT! When the print job (started directly from slicer) is canceled in Octoprint,
left panel displays after short time:
File: 3DBenchy_0.1mm_CPE_MK3S_3h10m.gcode 
Uploaded: 2020-02-03 20:31:15
User: API client
Timelapse: -
Filament (Tool 0): 4.28m / 10.29cm³
Approx. Total Print Time: 2.5 hours

So it looks like the analysis of received gcode file is not finished before the print is started?

fieldOfView, 2020-02-03:
> So it looks like the analysis of received gcode file is not finished before the print is started?

That is correct. If you upload a large gcode file, gcode analysis can take some time and resources to the point of possibly affecting sending gcode to the printer without hickups. So the background task of analyzing the gcode is stopped and postponed until the OctoPrint is no longer printing.

ochm, 2020-02-03:
> That is correct. If you upload a large gcode file, gcode analysis can take some time and resources to the point of possibly affecting sending gcode to the printer without hickups. So the background task of analyzing the gcode is stopped and postponed until the OctoPrint is no longer printing.

So it is not bug, but feature? Because it is confusig, and IMHO not olny for me. Or is planed some option to start API print after complete analyze is done?

foosel, 2020-02-03:
It is neither a bug nor a feature, it's a case of working with the available resources. Gcode analysis takes too many resources to be fun while a print is running. Postponing a print start automatically until analysis has run would confuse and about people, making it an option so far has not been on the agenda, could probably be implemented via a plugin though. Slicers can send analysis data along with the gcode file or wait for the metadata event to start with a print.

Closing as working as intended and explained before.

closed 2020-02-03 by foosel

ochm, 2020-02-04:
IMHO the RPi4 have enough resources, but OK.

fieldOfView, 2020-02-04:
CPython has something called the "Global Interpreter Lock",  which prevents multithreaded CPython programs from taking full advantage of multiprocessor systems in certain situations. This means that gcode analysis and USB printing would still be running on a single core, one intermittently waiting for the other.