OctoPrint/OctoPrint #3443: `SD init fail   ok P15 B3`

ipatch, 2020-02-04:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

_Honestly, not sure, this issue is more less related about SD card initialization_ 

#### What happened instead?

_I have an SD card inserted into a Big Tree Tech / BigTreeTech TFT35v2 that I use to print `.gcode` files from connected to a SKR v1.3 running marlin 2.x bugfix branch.

When I connect to the same board using octoprint `OctoPrint 1.4.0rc3` and successfully connect to the printer / board inside the printer, I see a message flash up on the BTT TFT35v2 with the following dialog in a "alert box"

```
SD init fail   ok P15 B3
```

[image]

This leads me to believe octoprint is trying to access the contents of the SD card because it is plugged into the SD card slot of the BTT TFT35v2.  Is it safe to assume that only one device can **mount** / access the contents of an SD card at a time? Or should octoprint be able to access the contents of the sd card while it's plugged into the touchscreen slot and be able to print a file `.gcode` from the SD card while it's plugged into the touch screen.

I don't have a deep understanding of octoprint or marlin or the BTT TFT35v2, but have worked on and off with computers for some time, so that's where my assumption about accessing the contents of the sd card come from.

And if that is correct, should octoprint run through a check to see if the contents of the SD card / or the physical device _RAW_ device already are in use by another "process" (using that term loosely) and then not try and access the contents of the SD card on the initial connection routine?

#### Did the same happen when running OctoPrint in safe mode?

[image]

#### Version of OctoPrint

OctoPrint 1.4.0rc3

#### Operating System running OctoPrint

```shell
λ cat /etc/octopi_version                                                                   
cat: /etc/octopi_version: No such file or directory
```

Running Octoprint in a _venv_ on Raspbian

```shell
λ cat /etc/os-release                                                                       
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

#### Printer model & used firmware incl. version

- Creality CR10S-pro _with skr v1.3 board installed with a BTT TFT35v2 attached to the board
- marlin 2.0 bugfix branch

#### Browser and version of browser, operating system running browser

- browser - vivalid 2.10.1745.27 (Stable channel) (64-bit)
- macOS 10.13.6 _computer accessing octoprint server via web interface_

#### Link to octoprint.log

[octoprint.log](https://gist.github.com/ipatch/f5b3593914fbfd0114a2d36883c4badb)

#### Link to contents of terminal tab or serial.log

```
cat serial.log                                                                            0 < 12:42:05
2020-02-04 12:10:09,096 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2020-02-04 12:10:36,618 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2020-02-04 12:11:22,891 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
```

#### Link to contents of Javascript console in the browser

**N/A** can paste if required

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2020-02-04:
The way I understand the way these touch screens work is that they are an actual client that connects to the printer, similar to how pronterface would, or in this case octoprint. The sd card is part of that client, not connected to the board directly, but I could be completely wrong. I think when you print from its sd card it's like printing a gcode file form local storage in octoprint. Because of this, the sd card isn't seen by the firmware or octoprint.

foosel, 2020-02-07:
Please enable serial log as instructed in the currently disabled log and share it.

I'm pretty sure though that this is simply caused by OctoPrint trying to init the printer's SD card, which for some reason can't happen, and the firmware has the resulting error exposed in the UI for some reason. Disabling SD support in OctoPrint under Settings > Features should take care of that.

closed 2020-06-12 by foosel