OctoPrint/OctoPrint #3444: [BUG] Octoprint cancels a paused SD print

MarkMan0, 2020-02-06:
#### What were you doing?

I started a print from the SD card, via the LCD. Then I paused it from the LCD, and octoprint registered it as Cancel

1. Start SD print
2. Pause SD print - either LCD, or send M25
3. Octoprint goes to state Paused, and then Cancelled

#### What did you expect to happen?
I expected octoprint to go to Paused state, and stay there, optionally execute any scripts

#### What happened instead?
Octoprint went to paused state, then Cancelled. 
The cancel script was executed more than once.

Right now, the cancel script is: M117 CANCEL

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint

OctoPrint 1.3.12 running on OctoPi 0.18.0

#### Operating System running OctoPrint

OctoPrint 1.3.12 running on OctoPi 0.18.0

#### Printer model & used firmware incl. version
Custom printer, running Marlin 1.1.9

#### Browser and version of browser, operating system running browser
Chrome

#### Link to octoprint.log
https://pastebin.com/JhhEjiFB

#### Link to contents of terminal tab or serial.log
https://pastebin.com/aukk3EHw

#### Link to contents of Javascript console in the browser
I don't think it's needed, but will share on request

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

MarkMan0, 2020-02-06:
I think the cause of the problem is, that Octoprint goes to state "Pausing" it sends "M25", which causes an //action:pause response, and it goes to a loop. That's why the cancel script is executed more than once.
This is just a guess though.
Are there any workarounds to this?
I'm quite familiar with this part of Marlin, so I could change the firmware in some ways to better cooperate with Octoprint, for my printer at least

MarkMan0, 2020-02-07:
I think I found the problem.
In Marlin, I commented the line, which sends _Not SD printing_. The main problem is, that this line is sent while the print is paused, and octoprint registers it as cancel.

I propose a solution:
Another SD state: _SD Printing Paused_. For the firmware, it's easy to tell whether it's paused or not printing.

As to why the cancelled script is executed that many times, I have no idea.

This is an older version of Marlin, 1.1.9. Not sure if this has been solved in any way in 2.0.x, anybody any insight?

MarkMan0, 2020-02-07:
M600 also appears to be bugged? Is it just for me, or is this a known issue?

foosel, 2020-02-07:
Neither is a known issue. However I'm currently quite busy with the RC and so it might take a while until I get around to analyse this issue. It does sound a bug however that crept in upon trying to get the detection of SD prints to work that are started from the printer controller instead of through OctoPrint

MarkMan0, 2020-02-07:
I think I got around the M600 and pause things. . Instead of sending action:pause, I now send action:paused, which skips the pausing state. This way I can pause/resume both SD and Host prints, either from the browser or the LCD, which is awesome.
Not sure how things work in the latest Marlin, but I'm gonna write any solutions  - in the firmware, I find.
After all, M600 should be handled by the printer, and not OPrint, so sending paused/resumed is fine, same goes for SD pause, if issued via the LCD.

I'm also working on stall detection, and I need to pause the whole print, similar to M600. That's how I got to this whole thing.

Unfortunately, I have zero experience with python, so I can't look under the hood of OPrint to find the best solution for this problem, but the FW tweaks I'll post here.

MarkMan0, 2020-02-10:
So I've solved it more or less as I said:
1. M125-Park head: now sends __action:paused__
2. M24 - Resume SD print: sends __action:resumed__
3. If the printer is moving planner.movesplanned() == TRUE and not printing from sd, I send __action:pause__, so OPrint will handle the pause.
4. If no moves planned, I can optionally send __action:resume__, to signal OPrint to continue sending gcodes.

This solved half of my problems, and this wasn't an OPrint bug IMHO, just miscommunication between the two developer parties.

But the cancel script is still executed many times, when I cancel an SD print, through the LCD. 
If needed, I can send logs.

wpegden, 2020-02-20:
I just wanted to comment on a related issue.  If I start a print from the LCD and THEN connect octoprint to the printer later, octoprint does not recognize that the printer is printing from the SD card, even though this is obvious from the terminal output.  

Should I start a separate ticket with logs, etc., for this issue?

foosel, 2020-04-07:
> Should I start a separate ticket with logs, etc., for this issue?

Yes.

foosel, 2020-04-07:
> Instead of sending action:pause, I now send action:paused, which skips the pausing state.

That's actually what should be getting sent. `//action:pause` should only be sent to instruct the host to pause, not if the pause is already being handled or if it actually originated from the host even. 

Usually `M600` does send neither though from my experience and just goes into `busy`, which is fine too. 

I'd consider this a firmware bug (sending the wrong action command, the one that instructs the host to initiate a pause instead of the one that tells the host a pause has been initiated) and am thus going to close this now. I'd rather not have to introduce yet another state into the state machine to handle these kind of behaviour issues (it's complicated enough as it is) when a firmware fix is actually the correct response.

closed 2020-04-07 by foosel