OctoPrint/OctoPrint #3445: Marlin doesn't 'stop' when cancelled from Octoprint

drewzh, 2020-02-06:
#### What were you doing?

1) Start print
2) Cancel print at any point

Notice that whilst Octoprint successfully cancels and Marlin no longer prints the incoming gcode stream, the printer remains in it's printing state and just idles, oozing hot filament with both heaters still going until 'Stop' in explicitly called via the LCD interface.

I've searched the current issues and any other similar issues people have faced. I've tried disbaled the logging of positions etc on cancel in the serial communications tab.

I've also enabled emergency parser in Marlin and disabled all TX/RX buffers to negate any buffering issues.

#### What did you expect to happen?
I expect the state of the printer to change to 'stopped' and the heaters to cool down.

#### What happened instead?
The printer remained in the 'printing' state at the last X/Y coordinates with both heaters going/nozzle oozing.

#### Did the same happen when running OctoPrint in safe mode?

Yes, same issue with Safe Mode booted

#### Version of OctoPrint

OctoPrint 1.3.12

#### Operating System running OctoPrint

OctoPi 0.16.0 H

#### Printer model & used firmware incl. version

Creality Ender 3 Pro with SKR 1.3 running Marlin 2.0.x (latest commit from today - issue has been apparent for many months of Marlin/Octoprint updates though - I've just 'lived with it')

#### Browser and version of browser, operating system running browser

Chrome 79.0.3945.130

#### Link to octoprint.log

https://gist.github.com/drewzh/7bf92030864df02b3d6fccb4b0ef0c5e

#### Link to contents of terminal tab or serial.log

https://gist.github.com/drewzh/7bf92030864df02b3d6fccb4b0ef0c5e

#### Link to contents of Javascript console in the browser

N/A - no warnings or errors logged. Just initialisation routines.

#### Screenshot(s)/video(s) showing the problem:

N/A - can provide, but I don't think it's required here

I have read the FAQ.

jneilliii, 2020-02-06:
Look in your GCODE Scripts section and make sure you have something in there to turn stuff off. Mine looks like this, which I believe is the default setting because I don't remember putting anything in there manually.

```
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0
```

foosel, 2020-02-06:
I agree, that sounds like missing cancel gcode, the snippet quoted by @jneilliii is shipped by default.

Please check if it's still set on your installation @drewzh and if not make sure to fix that. If it's set well have to dig deeper.

drewzh, 2020-02-06:
It sure looks like that was the issue. I'm so happy it's something so simple, thanks for pointing out my mistake! It's been annoying me for so long now, thinking it was an obscure bug. I should have reached out sooner! Thanks again! ❤️

closed 2020-02-06 by drewzh

Dloranger, 2020-11-25:
I too have been living with this issue in a slightly different form,  the bed turns off, but the extruder doesn't.  I have to manually kill the temperature on the temp page or reset the printer controller.

Not a show stopper, but not a good thing either.  I checked g-code scripts and they are set to the defaults shown above

foosel, 2020-11-25:
@Dloranger open a new bug report please, fully fill out the template and include a `serial.log`

Salamandar, 2021-05-19:
Hello,
I got the same issue, and i'm pretty sure the solution will work for me.

Although, I commented the "disable" snippets because I just wanted to keep the heaters running after a cancelling (to prevent having to wait *again* for preheatinng).
Is there a way to achieve that ? 

Oh found something (for people having the same issue as me) : https://docs.octoprint.org/en/master/features/gcode_scripts.html talks about `cancel_temperature` that may do the trick (power off heaters then power then on right after).