OctoPrint/OctoPrint #3448: Printer keeps request line 13 again and again, communication stuck at start of print - Anycubic I3 Mega

Zolarayze, 2020-02-09:
When trying to print a file that I uploaded to the SD-card on the OctoPi, it waits for the right temperatures and then "starts" to print. However, right at the start it stops and gives me the error: "Printer keeps requesting line 13 again and again, communicaiton stuck" and switches to offline (target temps back to 0 and no start of the print).

The printer I have is a 32-bit Anycubic I3 Mega. I saw there were some problems with this specific kind of printer and octoprint on #2309 and upgraded the firmware using the update.zw file and installed both .py scripts that were mentioned there. My firmware version now is V1.4.1RC2/P3. The scripts seem to be loaded as seen in the octoprint.log file.

If anyone can help me out, it would be greatly appreciated. I attached the serial and octoprint.log files.
[octoprint.log](https://github.com/foosel/OctoPrint/files/4177015/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/4177016/serial.log)

foosel, 2020-02-10:
Your printers manufacturer decided to ship it with firmware that had been broken by either then or someone else. It sends non standard responses that break the communication protocol and that is the issue here.

There are two plugins available that you can install to fix this but I also strongly recommend to complain to your printer vendor to fix their broken ass firmware. I can't do that for you.

See here for details: https://github.com/foosel/OctoPrint/issues/2309#issuecomment-358713557

closed 2020-02-10 by foosel

Zolarayze, 2020-02-11:
Hey @foosel thanks for responding, I saw the thread and installed the 2 scripts beforehand (I wrote about this in the main post under printer model and firmware.) . Unfortunately, it didn't fix the issue I am experiencing. It seems a little bit like the problem apos was talking about in #2309, however I tried that solution as well, but without luck.

Zolarayze, 2020-02-12:
Alright so it seems like the problem was my starting .gcode that I set in Cura.
Figured this out by receiving a different .gcode file that someone send me.
There were 2 commands on a single line.
I also changed Z endpoint to a new line.

`G28 X0 Y0 Z0 ;move X/Y to min endstops`
to
`G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops`

and 

`G0 Z0.15 G92 E0 ;zero the extruded length`
to
`G0 Z0.15 ;Drop to bed
G92 E0 ;zero the extruded length`

So if anyone else has this problem, check if there are multiple commands on the same line.