OctoPrint/OctoPrint #3451: print from sd card but transfer file to mounted sd card instead of over serial.

fractalfederation, 2020-02-15:
i just got my printer's sd card mounted, and did a custom upload_command for platformio so it would upload firmware to octopi > printer's mounted sd, then send m997 commend to reboot. its a lot faster than serial. was thinking about trying to make print from sd transfer to mounted sd instead of over serial.

was curious if there is already a hidden feature or something?

wpegden, 2020-02-15:
This is a great idea.  I use a flashair card on the printer and it is visible locally on the machine the octoprint servers is on. Would be great to upload directly to it through the octoprint interface.

jneilliii, 2020-02-15:
If it's available via terminal couldn't you create a mount point as a subfolder to the uploads folder then upload to that sub folder in octoprint and then refresh the sd card file list? I vaguely remember a plugin that would redirect sd card upload to flash aircard. You might want to search the community forum.

fractalfederation, 2020-02-16:
@jneilliii ill try that

jneilliii, 2020-02-16:
Here's one of the discussions.

https://community.octoprint.org/t/wifi-sd-card-flashair-integration-in-octoprint-working-code/15216