OctoPrint/OctoPrint #3453: WiFi AC issue with raspberry Pi 4

Guilouz, 2020-02-16:
Hi,
Is Octoprint only compatible with wifi N ?

The new Raspberry Pi 4 is WiFi AC compatible but it only connects with WiFi N with Octoprint.

I use OctoPrint 1.4.0rc5 with Octopi 0.17.0

See :
[image]

[image]

tedder, 2020-02-16:
It's likely this is in [OctoPi](https://github.com/guysoft/OctoPi), but note Pi 3B+ also supported 5ghz. If you're talking about 802.11n on 2.4ghz, [all Pis with built-in wireless support it](https://www.raspberrypi.org/documentation/faqs/#hardware). So you'll probably need to do some troubleshooting to figure out where it broke down. It's probably better discussed [on the Octoprint community forum](https://community.octoprint.org/latest) (or a Raspbian site) than in a bug ticket.

foosel, 2020-02-17:
OctoPrint doesn't know anything about WiFi, it's a user space program.

I suggest to get in touch on the OctoPrint community forum where someone using OctoPi as well might have some hints for you.

closed 2020-02-17 by foosel