OctoPrint/OctoPrint #3454: Python 3 will break some plugins

gdombiak, 2020-02-16:
#### What were you doing?

Hi @foosel, I while back I opened a ticket about some _str vs bytes_ issue I ran into while using CancelObject plugin with OctoPrint 1.4 with Python 3. I then closed that ticket assuming the issue was with the plugin but I now think that we have a Python 3 vs Python 2 difference that will break some plugins.

1. Put a breakpoint in DiskFileWrapper#save and/or StreamWrapper#save
1. Upload new file to OctoPrint
1. Execute _io.BufferedReader(self.stream()).readline()_ from debugger
1. Python 3 returns **bytes** while Python 2 returns **str**
1. This will break plugins that have a _process_line_ hook like DisplayLayerProgress or CancelObject. See LineProcessorStream#read when process line is called. The line parameter used to be of type str with Python 2 and is now bytes with Python 3.

An example of a Plugin breaking is here: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/117. The hard part is that there is no exception but a _semantic error_. :(

#### What did you expect to happen?

line parameter passed in LineProcessorStream#read to be of type str so existing plugins that have a hook do not break. Otherwise each plugin will need to handle str and bytes

#### What happened instead?

Functionality of plugins break

#### Did the same happen when running OctoPrint in safe mode?

In safe mode the error is not exposed. OctoPrint without plugins also works fine but problem is waiting to unleash its fury against poor unaware plugins. :)

#### Version of OctoPrint

Any OctoPrint version that runs with Python 3

#### Operating System running OctoPrint

Python 3

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

Python 2
[image]

Python 3
[image]

I have read the FAQ.

foosel, 2020-02-18:
So I took a close, long and hard look at this, and came to the conclusion that I cannot do anything about this on my end.

The thing is, the class in question implements `io.RawBaseIO`, which means I have to adhere to its API. Which means that it needs to operate on bytes (`str` in Python 2, `bytes` in Python 3), not unicode (`unicode` in Python 2, `str` in Python 3). If I start decoding stuff into unicode before passing it to `process_line` I get into all kinds of troubles when further processing the result from that inside `read`. I could turn it back into bytes, but that just screams trouble too...

The thing is, if you work on the file system level, you work with bytes in Python. You always have, their data type was just called `str` in Python 2. Client code needs to recognize that. I try to make the py2/3 compat migration as painless as possible for plugins, but that's something I *cannot* do for them without breaking parts of OctoPrint.

So I'm leaving this as is but added some warnings to the code which should make it into the documentation. I'll also finally look into whipping up some kind of list of common pitfalls for plugin authors to refer to for the coming months of migrations. Should have done that already but other stuff was even more urgent.

closed 2020-02-18 by foosel

gdombiak, 2020-02-21:
Thanks Gina @foosel for looking into this. Your approach makes sense to me. Having documentation providing guidance about the problem and possible solutions that plugin developers should implement is going to be helpful to many people.

Regards,
Gaston