OctoPrint/OctoPrint #3457: Octoprint leaves printer LCD state as Printing...

Kokorone, 2020-02-18:
#### What were you doing?

1. Start a print with Octoprint
2. Allow the print to finish, or cancel it
3. Observe the Octoprint status switch back to "Operational"
4. Physically look at the printer LCD and see that it is showing "Printing..." with a timer counting up as well as a menu option to "Cancel print"

#### What did you expect to happen?

Printer LCD to return to normal idle state as if nothing is printing

#### What happened instead?

Remained in Printing... state

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

Octoprint 1.4.0rc5

#### Operating System running OctoPrint

Octopi 0.17.0

#### Printer model & used firmware incl. version

Creality CR-10 w/ SKR 1.3 running latest marlin bugfix 2.0.x as of 2/18/2020

#### Browser and version of browser, operating system running browser

Chrome 80.0.3987.106, Windows 10

#### Link to octoprint.log

https://pastebin.com/ne0Jv2kR

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-02-19:
OctoPrint doesn't send a "Printing..." text to your printer's LCD. That's probably part of your sliced GCODE or your configured GCODE scripts. If you want it to be set to something else at end or cancellation of a print,adjust the slicer and/or GCODE scripts in OctoPrint accordingly.

closed 2020-02-19 by foosel

Kokorone, 2020-02-19:
Rather than it being a bug with octoprint sending a “Printing...” text to the printer, it is a bug about Octoprint NOT sending the command to tell the printer timer to stop and that the print is complete which is why I submitted it as a bug.

foosel, 2020-02-19:
It can't possibly do that because there is no such standardized command ;) 

The only thing there is that's usually available is an M117 command to display arbitrary text on the printer's display. Some soccer's include a line `M117 Printing...` at the start of the sliced result. That's completely outside of OctoPrint's control though. If you want to change or modify this kind of behaviour, change your slicer profile or modify the gcode scripts configurable in OctoPrint to induce whatever M117 message you want to be sent in case of print success or failure.

Kokorone, 2020-02-19:
Ended up solving this issue on my own. I had to add commands for M104 S0 and M140 S0 into the gcode script for finished printed and canceled prints in Octoprint. Despite those already being in the end gcode in the slicer it didn’t signal the end of the print on the lcd until I added those to Octoprint.

wikolii, 2020-05-21:
I don't understand why was this issue closed.
I'm having the same problem. Besides the "printing" message on the LCD, the main issue is that the printer (not octoprint) is left in "printing" state. Showing the "pause" and "stop" print messages on the menu. Also, if successive print jobs are sent one after the other, the LCD will show the cumulative "print time" for all print jobs until reset or manually stopped from the LCD.

The workaround proposed to send the cooldown commands should not be a solution. When i cancel a job i usually intend to print it again, so cooling down will make me loose time if i forget to preheat manually.

Please reopen and validate, the "M117 Printing" code is not the issue here.

trikko, 2021-03-30:
This bug is still there and annoying.

I have to select "stop print" from screen to end it and do anything else.

foosel, 2021-03-30:
This "bug" is either caused by something you have in your gcode or by your printers firmware switching to some printing state entirely on its own. In either case it's not something that can be solved on OctoPrint's end.