OctoPrint/OctoPrint #3458: [Request] Add prevent_popup to MOVIE_DONE event payload

FormerLurker, 2020-02-20:
**Is your feature request related to a problem? Please describe.**
Several plugins rely on the MOVIE_DONE event to trigger actions, and I'd like to take advantage of that within Octolapse.  I used to call this event, but now Octolapse can store timelapses in a custom folder, so it's not guaranteed that a new timelapse video will be available to the stock timelapse plugin.  Additionally, since there is a new rendering progress display on the Octolapse tab, I removed the popup completely since Octolapse produces too many popups as it is.  So, if I send the complete event payload, a popup gets displayed, but it's possible that no timelapse will be available within the stock timelapse tab, which confuses users.

**Describe the solution you'd like**
If you can add a 'prevent_popup' payload option that prevents the stock timelapse plugin from showing a popup it would pretty much solve the issue.  It's still fine to to fetch data from the server, since it is possible that a new timelapse is available.  Currently I'm omitting the movie_prefix field, which prevents the popup at the cost of a javascript error.  I can add an OctoPrint version check and continue omitting movie_prefix until this feature is added.

**Describe alternatives you've considered**
Originally I added my own custom event, but now plugins designed to work with MOVIE_DONE aren't working anymore.  I'm working with some plugin developers to work around this by helping them add the custom Octolapse event, but this will be a never-ending process I'm afraid.

**Additional context**
NA

jslay88, 2020-02-21:
Parent-Child structure to events would be nice, but would require complete redesign/restructure of events and the event system moving over to something more like a pub/sub system; where a developer can trigger on parent or child events accordingly, only having a single event call.

FormerLurker, 2020-02-22:
@foosel, after a lot of thought, I don't think my original request needs to be implemented.  Bottom line that the stock timelapse MOVIE_DONE event, as is, will never quite do what I want it to do, because it means something different (a stock timelapse has completed).  Even if a more advanced solution is implemented (@jslay88's suggestion above about tagging the sender), it would still require specific programming for plugins to take advantage of this.  That would be a useful feature, for sure, but at the end of the day I think it's simpler to watch for Octolapse's custom event, which means exactly what I want it to mean (Octolapse is finished creating a timelapse).

Closed!

closed 2020-02-22 by FormerLurker