OctoPrint/OctoPrint #3459: 1.4.0rc* Login sessions not persistent even with "Remember me" selected.

ChrisHeerschap, 2020-02-21:
#### What were you doing?

That's the tricky part, I've been trying to figure out exactly what triggers it, but so far I have not been able to. I thought it was accessing it from multiple systems, but I've accessed it from two computers and three browsers simultaneously, and it's working.

I also access it remotely using SSH port forwarding, and thought that might contribute - since I'm accessing the same instance with two methods via the same browser - directly when I'm on my home network, and via localhost:8000 with ssh port forwarding when I'm remote. This doesn't seem to be it either as I'm currently viewing it directly _and_ via port forwarding in Chrome on my macbook while also accessing it via Firefox on that system and Chrome on an iMac. I also threw in accessing it from my iPhone and still can't get myself logged out.

To really try and trigger it, I connected to my older octoprint raspi which isn't currently connected to a printer, and connected to another printer I have at work via SSH port forwarding. Four tabs open in Chrome connected to three different octoprint instances (all some version of 1.4.0rc*) and I _still_ can't get it to happen!

So - I still haven't conclusively been able to nail down what causes it - which is why I haven't opened a ticket previously. I really wanted to nail down what was going on before opening a ticket but still haven't.

#### What did you expect to happen?

I would think if I click "Remember Me" I would stay logged in on a particular browser, preferably across browser restarts. Sorta like a gmail login - I would only have to login if I specifically logged out.

#### What happened instead?

I often find I have to login again at strange times without having chosen to log out.

#### Did the same happen when running OctoPrint in safe mode?

I ran in safe mode at one point and it didn't happen - but I honestly didn't run it long enough that I can conclusively say it _didn't_ happen at all.

#### Version of OctoPrint

1.4.0rc5 currently.

#### Operating System running OctoPrint

Raspbian installed via OctoPi 0.17.0 on a RPi 4. Previously Raspbian from OctoPi 0.15.0 (although I keep the raspbian up to date with apt as updates are available) but that one was running rc4.

#### Printer model & used firmware incl. version

Prusa Mk3S with latest 3.8.1 firmware

#### Browser and version of browser, operating system running browser

Chrome, latest, as well as Firefox, latest - all on Mac OS, latest on the macbook but the imac is way back at 10.11.

Also worth mentioning is I run uBlock Origin adblocker plugin on all browsers as well as Privacy Badger, and several more Chrome extensions, but I don't think any of those should be interfering. I have disabled privacy badger 

#### Link to octoprint.log

https://gist.github.com/ChrisHeerschap/c21786027faa9c328914c5320e487317

My most recent logout happened at `2020-02-20 19:28` and shows up in the logs. 

#### Link to contents of terminal tab or serial.log

didn't have serial logging enabled - and don't _think_ that should affect it.

#### Link to contents of Javascript console in the browser

https://gist.github.com/ChrisHeerschap/dd1cd8216b07c43238b2cb6b639c4e96

#### Screenshot(s)/video(s) showing the problem:

Don't think screenshots will help - it's either logged in or it's the login page. Can provide if they would provide some benefit.

I have read the FAQ.

Finally, yes, your German pronunciation sounds very close to the proper Dutch pronounciation - which a Dutch person told me that despite my best efforts, I wasn't saying it _quite_ right, so I usually go with the first way you pronounced it on the On the Air #29 but I don't get cranky about any variation considering I can't properly pronounce _my own name!_ 😄 

Thank you! I am not certain that this issue isn't coming from something I'm doing wrong or something special about my use case, so if nobody else is having this problem, I wouldn't hold off on releasing 1.4.0.

ChrisHeerschap, 2020-02-21:
Sorry to cause such stress! Am still not 100% sure this isn't unique to some dumb thing I'm doing.

ChrisHeerschap, 2020-02-21:
_IF_ I can make this happen again (been trying really hard with no luck) I will grab and post the javascript console logs immediately.

ChrisHeerschap, 2020-02-22:
Finally happened again when I checked this morning. Here's the series of steps:

* Was on my macbook air, octoprint tab was not opened. I opened it, and it had me logged in as expected. 
* Sent a wake on lan packet to the iMac as I was too lazy to go to it and login directly.
* Once it was responding on pings, I logged in via VNC and switched to the Chrome tab and it was logged out.
* Checked on my iPhone and it was still logged in and working.
* Closed the tab on my macbook air and reopened and it worked.

Fresh octoprint.log available here: 
https://gist.github.com/ChrisHeerschap/374a50359b4a94ba72b7ccb064ce8e14

ChrisHeerschap, 2020-02-22:
Looking through the settings, this seemed like it _might_ be useful on your end:

Usage tracking Instance ID: 00201d43-6a59-4c60-9319-5bcd3b05468a
Error tracking instance ID: 3e4fadff-c616-4fb1-812d-7b88feceb5f2

foosel, 2020-02-24:
I have two questions:

  1. You say on the MB Air it worked again after you just closed the tab and reopened it? I take it that means you didn't manually log in or anything, you simply closed and reopened the tab? Would reloading it also have worked?
  2. You say you were logged out on the MB Pro. Was that in an existing tab or a newly opened one?

Taking a look at the first log, I guess this is from a former session:

```
2020-02-21 18:44:54,749 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.47.117
2020-02-21 18:44:54,890 - octoprint.server.util.flask - INFO - Passively logging in user cmh from ::ffff:192.168.47.117
2020-02-21 18:44:54,938 - octoprint.server.util.sockjs - INFO - User cmh logged in on the socket from client ::ffff:192.168.47.117
[...]
```

and this is when you logged in via VNC and switched to the tab in question:

```
2020-02-21 20:45:54,530 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.47.117
2020-02-21 20:45:54,668 - octoprint.server.util.flask - INFO - Passively logging in user cmh from ::ffff:192.168.47.117
2020-02-21 20:45:54,669 - octoprint.server.util.sockjs - INFO - User cmh logged out, logging out on socket
2020-02-21 20:45:54,743 - octoprint.server.util.sockjs - INFO - User cmh logged in on the socket from client ::ffff:192.168.47.117
2020-02-21 20:45:55,375 - tornado.access - WARNING - 403 GET /api/settings (::ffff:192.168.47.111) 86.65ms
```

For some reason it runs through the logout code there, which I'll have to dive in to how it can happen, considering that right after it reports you as logged in on the socket again. And a couple seconds later I take it this is the closing/reopening of the tab:

```
2020-02-21 20:45:55,396 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.47.111
2020-02-21 20:45:57,596 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.47.111
[...]
2020-02-21 22:46:45,622 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.47.117
2020-02-21 22:46:45,766 - octoprint.server.util.flask - INFO - Passively logging in user cmh from ::ffff:192.168.47.117
2020-02-21 22:46:45,814 - octoprint.server.util.sockjs - INFO - User cmh logged in on the socket from client ::ffff:192.168.47.117
```

foosel, 2020-02-24:
Good news, I think I've figured out what's going on, and it's unrelated to the remember me cookie and rather a result of stale sessions being purged from the system and a new logout propagation mechanism under 1.4.0 that means the browser will actively log out when that happens (instead of staying in a semi logged in state as before, which causes a different set of issues).

The proper solution here would be to require a reauthentication instead of doing a logout, and that's what I'll have to look into now.

Long story short is that the login session gets cleaned up 24h after its creation (it should probably rather be 24h after last use too). That now triggers a message on the socket (which is where the above logging comes from) and that message in turn causes the browser to log out as well. Which is why I couldn't reproduce this on purpose, it needs a stale login session to happen. I just set the timeout to 1min instead of 24h and could reproduce though.

foosel, 2020-02-24:
Should be fixed by the above commit, ready for 1.4.0rc6

CapnBry, 2020-02-25:
Oh boy I was just coming to report this exact thing. I have had it happen multiple times where I load the webui in a browser, I don't have to login, everything works, and leave the window open while doing things and I look back over and I'm looking at a login prompt. I'd been trying to track it down for days now since it rarely happened. I could tell from logging I had turned on that the browser seemed to be actively requesting a logout which had me greatly puzzled.

I'll pull the latest code and see if it fixes this issue.

foosel, 2020-02-26:
1.4.0rc6 is out

closed 2020-02-26 by foosel

ChrisHeerschap, 2020-02-26:
Awesome, thank you so much! As soon as this print finishes I'll update!

commit 249d19f5 (2020-02-24): 🐛 Reauth for stale sessions instead of active logout

Fixes #3459