OctoPrint/OctoPrint #3460: [Request] Increase maximum number of extruders

NielschB, 2020-02-23:
**Is your feature request related to a problem? Please describe.**
Currently I am developping a multi material unit for my prusa MK3 which can handel up to 18 filaments. Octoprint number of extruders is limited to 10, where I would require 18.

**Describe the solution you'd like**
I would like to have the maximum number of extruder increased to at least 20.

**Describe alternatives you've considered**
No altrenatives at the moment

**Additional context**
No additional context needed.

cp2004, 2021-07-16:
OctoPrint supports any number of extruders, the only thing that is stopping you is the red highlight when you enter more than 10 in the box. This restriction has been removed for 1.6.0, see #4149

cp2004, 2021-10-11:
This was implemented in 1.7.0 which has been released 🎉