OctoPrint/OctoPrint #3463: [RC Feedback] Feedback on the 1.4.0rc6 Release Candidate

foosel, 2020-02-26:
Please provide *general* feedback on your experience with the 1.4.0rc6 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

  * ❓ #3394 comm port autodetection still broken - appears to not be a regression and will get a closer look in 1.4.1

### Unrelated third party issues

### Issues needing analysis

schnello, 2020-03-01:
RC 6 installed without problems. Also prints are fine and no issues.

ChrisHeerschap, 2020-03-02:
Installed on two raspi systems as soon as I heard it was out (well, one print had to finish first) but so far everything is looking roses. Have been accessing it from many different systems (admittedly all apple) concurrently and it's looking good. No recurrence of the logout issue as of yet, have one system i use infrequently that I'm giving a couple days before I test with that one again. Everything else has been flawless.

ChrisHeerschap, 2020-03-04:
Wuhu! 1.4.0 is out!  😃

foosel, 2020-03-04:
Oh right! *closes this* ^^

closed 2020-03-04 by foosel