OctoPrint/OctoPrint #3464: Octoprint crash mid print

JustAnother1, 2020-02-27:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

I was printing a big (45Mb) gcode file. It was running OK for 16 or so hours. No issues, Web Stream OK. Evrything fine.

#### What did you expect to happen?

the print should continue for a few hours.

#### What happened instead?

It then suddenly stopped printing. Nozzle stopped moving and Octoprint web Interface switched to "Server is offline"
After a Relaod I get a page with OctoPrint server is currently not running with instructions to check log file and restart server.

Restaring brings the server back

#### Did the same happen when running OctoPrint in safe mode?

It did happen only this once, therefore I could not test in safe mode.

#### Version of OctoPrint

OctoPrint 1.3.12 on OctoPi 0.15.1

#### Operating System running OctoPrint

OctoPi 0.15.1 on a RaspberryPi 3

#### Printer model & used firmware incl. version

Ultimaker Original with Heated Bed upgrade.

#### Browser and version of browser, operating system running browser

Firefox 73.0.1 (64bit) on Linux Mint 19.3

#### Link to octoprint.log

<script src="https://gist.github.com/JustAnother1/437f6bad6782fdc589b9996d4a8f8b45.js"></script>

#### Link to contents of terminal tab or serial.log

Octoprint crashed so no access to terminal tab, sorry.

#### Link to contents of Javascript console in the browser

Did the restart before noticing you might need that. sorry.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-02-28:
Sadly the only thing hinting at a crash in the log is this loooong line of zero bytes between 22:34 and 22:35 on February 26th.

Was that the crash you mentioned?

If the process is actually gone it could be something crashing at such a low level that the log didn't even catch it. In that case running OctoPrint from terminal (maybe in something like a `screen` instance) and trying to reproduce might be an approach. If it can be reproduced and wasn't just an one-of-thing 😕

JustAnother1, 2020-02-28:
No the crash happened shortly before my report so Feb 27th on 17:something.

I did the show python process command on that instructions site (ps | grep something and it showed no processes.

JustAnother1, 2020-03-01:
I redid the print with octoprint started in a screen. But the print succeeded. I don't know if I should be happy or sad about that.

 I close this issue because I can not reproduce it. Thanks for looking into it!

closed 2020-03-01 by JustAnother1