OctoPrint/OctoPrint #3465: Redirect Stream to other URL

nicedevil007, 2020-02-27:
Hey all out there :)

I like the included option to stream a video of my prints but I have a little problem with it.
I'm using a Sophos UTM Firewall in my home network and use it as reverseproxy as well.

Unfurtunately I'm not able to autoredirect the incoming traffic on my dns like
webcam.nicedevil.io to webcam.nicedevil.io/webcam/?action=stream.

That is because of the questionmark in this type of URL.

Now I just want to ask if it is possible right now to change the used URL for the stream or add this as a new feature to octoprint?

Thank you all in advance.

foosel, 2020-02-28:
The webcam server is not part of OctoPrint but rather mjpg-streamer which comes bundled with OctoPi (see https://community.octoprint.org/t/what-is-the-difference-between-octoprint-and-octopi-are-they-the-same-thing/185). 

Your request also sounds like something better suited for [the community forum](https://community.octoprint.org) rather than the issue tracker of either project.

closed 2020-02-28 by foosel

nicedevil007, 2020-03-07:
I found a solution with a friends help now. Just in case anyone is trying the same to achieve.

Change the settings of OctoPi's haproxy (/etc/haproxy/haproxy.cfg)

```
backend webcam
        acl p_ext_jpg path_end -i ?action=stream
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
        http-request set-path /?action=stream if !p_ext_jpg
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http
```

1st and 4th of this config is what we need.
Then if someone is going on https://urltoOctoPrint/webcam it will redirect to https://urltoOctoPrint/webcam/?action=stream

In my case I was able to permanently redirect all traffice to the main URL to /webcam and then it showing the stream.