OctoPrint/OctoPrint #3467: Temp graph and temp UI not refreshing after selecting Tool 1 in shared nozzle configuration

MoffTigriss, 2020-03-01:
#### What were you doing?

Printing with a dual extruder and a shared nozzle, on a Klipper printer (Voron 2.1).

When using T0, the temperature graph is correct. When using T1, tool temp is frozen, until reusing T0.

This is not an issue for printing, but UI is not working properly, and third party tools using API are fooled too (OctoScreen, in this case). The Klipper-controled LCD screen is all good too.

1. In the printer profile, "Hotend & extruder", change number of extruders to 2, then activate "Shared nozzle", Save.
2. Control > Select tool > Tool 0
3. Temperature -> set to 50°C, wait for stabilization
4. Control > Select tool > Tool 1
5. Temperature -> set to 60°C, wait a little
 Graph is not updated, temp is frozen, and the Target input is not updated.
6. Control > Select tool > Tool 1
7. Temp is now read again, the Target input is correct with the last input (60°C), graph is updated again.

#### What did you expect to happen?

Temp, graph and input showing current temp / set for when Tool 1 is selected in a shared nozzle configuration.

#### What happened instead?

Freezing of those information when selecting Tool 1.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.4.0rc6 (bug present in last stable)

#### Operating System running OctoPrint

OctoPi 0.16.0

Tested on a fresh install too.

#### Printer model & used firmware incl. version

Voron 2.1, Klipper firmware, last commit (bug present since my first try in december)

#### Browser and version of browser, operating system running browser

Firefox, Chrome. Windows and Linux. Tested with Firefox Android too.

#### Link to octoprint.log

https://gist.github.com/MoffTigriss/cc1350b42a2b4946434c1e0e751f5571

#### Link to contents of terminal tab or serial.log

https://gist.github.com/MoffTigriss/811c84d03035b6cc698b4b463b209dbb

Not for that log exactly, but terminal messages are totally ok.

#### Link to contents of Javascript console in the browser

https://gist.github.com/MoffTigriss/0ec9ed2929fc93c4a00cc382f05e9609

#### Screenshot(s)/video(s) showing the problem:

https://tof.cx/image/qQngr1

The jump from 55 to 65 is just after switching to T0 again.
I hope my report can be useful :)

I have read the FAQ.

Sp4rky001, 2020-04-04:
I'd like to add that I'm facing this issue as well on Geeetech A10M with a shared nozzle.  I believe it's because the M105 command is sending back T0 temperatures only and from what I gather from a previously logged issue, Octoprint may be looking for the response to send back T:<temp> when shared nozzle.

This is the M105 response from Klipper running on my Geeetech A10M:

> Send: M105
Recv: ok B:22.8 /0.0 T0:47.6 /0.0

jneilliii, 2020-06-07:
I think @b-morgan is working on a possible plugin solution for this.

foosel, 2020-06-12:
Possible for solution for this as an intermediary is a good idea, but this is in fact a bug in the temperature parser. I didn't anticipate a combination of values that is intuitively valid. Working on a fix for 1.4.1.

commit 0a9e9ef3 (2020-06-12): 🐛 Fix handling of shared nozzle reporting w/ only T0

Fixes #3467
closed 2020-08-04 by foosel