OctoPrint/OctoPrint #3468: Multiple instances of OctoPrint don't play nice besides eachother anymore

opiswahn, 2020-03-01:
With the latest RC (i belive rc5 also had this problem) there are issues with running multiple instance of octoprint on the same Raspberry Pi.
With the current stable version 1.13.12 everything runs as expected, but with  the current RC, one instance is running fine, but as soon as another one is activated(0start to print)m i get serial exception errors on all instances, either the instances stop with error and decoupple from the printer, or printing behaves very strange, resulting is printing only spaghetti.

#### Version of OctoPrint

#### Operating System running OctoPrint

OctoPi on Raspberry Pi 3 (0.15.1)

#### Printer model & used firmware incl. version

Printers running Marlin 2.0.x

tedder, 2020-03-01:
Multiple instances on one instance is not supported, but there is a thread for discussion.
https://community.octoprint.org/t/multiple-printers-per-server-megathread/7778

opiswahn, 2020-03-05:
Damn, i can't belive it, it was the power supply failing... caused fluctuations in the supply that must have some effect on the chips on the controller board...
I replaced it, now everything is again working as expected...

closed 2020-03-05 by opiswahn