OctoPrint/OctoPrint #3469: [Request] adding a function to distinguish between heated chamber, autochamberfan and only a "chamber"/room tempsensor

docgalaxyblock, 2020-03-03:
Hi, I think it would be great if there was a checkbox to switch between a heatable chamber and only a sensored chamber/room.
Or even better to have an additinal checkbox were you can select that you have an automatic chamber fan

I think this says it all in a nutshell

fieldOfView, 2020-03-04:
> I think this says it all in a nutshell

I don't think it does. Apart from showing those two checkboxes, what would this do? Would anything change in the OctoPrint interface or how the temperatures are shown in the graph?

foosel, 2020-03-04:
Agreed, this really does not "say it all in a nutshell". More details please. What where why.

docgalaxyblock, 2020-03-07:
@foosel Request updated

jneilliii, 2022-01-23:
So I think my PlotlyTempGraph plugin can handle all these requests of labeling, hiding/showing temps, etc. closing.

closed 2022-01-23 by jneilliii