OctoPrint/OctoPrint #3471: Confusing dialogue when cancelling print

FilmMartin, 2020-03-04:
When cancelling an active print (pushing the Cancel button), the following dialogue appears:

> This will cancel your print.
> Are you sure you want to proceed? (Options: yes / no)

Pushing "yes" does cancel the print, while "no" resumes it. This is not exactly intuitive.

It would be better to simplify the dialogue to

> Do you want to cancel your print
> Options: No, resume // Yes, cancel  

Thanks for considering :)

FilmMartin, 2020-03-04:
P.S. This concerns the file "messages.io" and these lines, I don´t know enough about Github to suggest the changes directly. sorry.

#: src/octoprint/plugins/softwareupdate/static/js/softwareupdate.js:528
#: src/octoprint/static/js/app/helpers.js:736
msgid "Are you sure you want to proceed?"
msgstr "Bist Du sicher, dass du fortfahren willst?"

tedder, 2020-03-05:
"are you sure you want to proceed"
"yes I want to proceed"

seems to work for me.  "yes, cancel" seems like an improvement, but does that mean "I wish to cancel canceling" or "I wish to cancel the print"? "resume" isn't as problematic.

FilmMartin, 2020-03-05:
The first time I saw that dialogue, I was not sure how to interpret it; "Do you want to proceed" could either mean "Do you want to proceed with the print (i.e. not cancel)" or "Do you want to proceed with canceling (i.e. cancel)."

Not a biggie for me anymore, as I have learned it now, but for new users it might be a hiccup.

foosel, 2020-03-05:
How about:

[image]

FilmMartin, 2020-03-05:
Perfect. Vielen Dank :)

foosel, 2020-03-05:
Implemented by the above commit, to be released with 1.4.1

commit 9781d5c1 (2020-04-07): 🚸 Rewording of buttons in cancel confirmation

Closes #3471
closed 2020-08-04 by foosel