OctoPrint/OctoPrint #3472: Update on Octoprint 1.4.0 failed due to out of memory...

energiebrand, 2020-03-05:
I trried a lot of times to update to 1.4.0 but always the same failurue. I´m not an expert of linux! I am running a RPi 3B+ on a 16GB SD card. Attached some screenshots of my memory (df -h). The requested logs and the failure message. 
I also reduced the memory split to 16MB (no GUI use). Now I am running out of ideas. Maybe someone can help what to do.
[image]
[image]
[image]
[plugin_softwareupdate_console.log](https://github.com/foosel/OctoPrint/files/4292976/plugin_softwareupdate_console.log)
[octoprint.log](https://github.com/foosel/OctoPrint/files/4292977/octoprint.log)

foosel, 2020-03-05:
`pip` by default will try to download the update first to `/tmp`. Which on your system is only 8MB in size and hence will definitely not suffice to download *& unpack*  the update. Hence the error. OctoPi doesn't ship like this, so I'm assuming this is a custom install? I suggest to remove the `/tmp` mountpoint from `/etc/fstab` again and have `/tmp´ be on the root partition, which has plenty of space.

In general, for questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2020-03-05 by foosel

energiebrand, 2020-03-05:
OK, thanks a lot and sorry to choose the wrong channel for my problem. Will try to follow your recommendation!

energiebrand, 2020-03-05:
Solved it by your advice! Thank you! (fstab now 100MB)