OctoPrint/OctoPrint #3474: 1.4.0 Temperature graph not working

bluemanuk, 2020-03-06:
[image]

Upgraded to 1.4.0 and I no longer get any temperature graph and can not control the temperatures from the UI. I can still print and I can see the temperature coming through the terminal window. 

Running a CR10s

Have rebooted the printer and Octoprint several times to see if that clears the problem but no luck

Log file attached. 
[octoprint.log.2020-03-05.txt](https://github.com/foosel/OctoPrint/files/4297901/octoprint.log.2020-03-05.txt)

bluemanuk, 2020-03-06:
#### What were you doing?
Upgraded to octoprint 1.4.0

#### What did you expect to happen?
Temperature graph to show as normal. 

#### What happened instead?
No graph data and unable to adjust temperatures on the screen.

#### Did the same happen when running OctoPrint in safe mode?
No 

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

bluemanuk, 2020-03-06:
issue to do with TP Link plugin.

closed 2020-03-06 by bluemanuk

hbh7, 2020-03-21:
I'm having the same issue but don't have that plugin. How did you figure out which it was? Hopefully not by manually going thru and disabling them one by one, right? lol

bluemanuk, 2020-03-21:
> Reply to: foosel/OctoPrint <reply@reply.github.com>

Hi,

 

To be honest I think it’s octoprint. I have found that I need to either re-start octoprint or go into safe mode and then boot normally to get it back. It seems to have no correlation to any specific plugin.

hbh7, 2020-03-27:
Figured it out for me. It was the OctoPrint-Tempsgraph plugin. Didn't realize I had installed that, otherwise I guess it should have been more obvious lol. Time to open an issue on their end