OctoPrint/OctoPrint #3475: get_api_connection state reporting "perational"

empakoso, 2020-03-06:
FYI, anyone else getting this response:

OctoPrint state is unkown
2020-03-06 18:10:01,569 INFO     GET_API_connection state From OctoPi
2020-03-06 18:10:02,292 INFO     GET_API_connection attempt: 1 @ API response: 200
2020-03-06 18:10:02,293 INFO     GET_API_connection received successfully
2020-03-06 18:10:02,294 INFO     parsing connection response
2020-03-06 18:10:02,294 INFO     Current connection state: perational attempt no.: 0
2020-03-06 18:10:02,885 INFO     OctoPrint state is unkown to PiConn.py
2020-03-06 18:10:03,286 INFO     GET_API_connection state From OctoPi
2020-03-06 18:10:03,942 INFO     GET_API_connection attempt: 1 @ API response: 200
2020-03-06 18:10:03,943 INFO     GET_API_connection received successfully
2020-03-06 18:10:03,944 INFO     parsing connection response
2020-03-06 18:10:03,944 INFO     Current connection state: perational attempt no.: 0

tedder, 2020-03-06:
There is no file in this release called `PiConn.py`. What plugins do you have? Please upload a **complete** log file.

empakoso, 2020-03-07:
Wait, my mistake.  It’s a parsing error on my part.
Sorry fella’s

closed 2020-03-07 by empakoso