OctoPrint/OctoPrint #3476: Ability to disable M27 messages altogether

frankhommers, 2020-03-07:
When I am in the file section menu of the Ultimaker 2 control panel, and I haven't confirmed the file, but I'm scrolling over it. The UM2 sends out the filename (which I am hovering) to OctoPrint.
After that OctoPrint starts sending M27.
After that I can't select any other file.

The M27 messages make my hardware control unusable. I commented it out in the code and then everything works fine. It would be great to make it an option to disable M27 polling.

frankhommers, 2020-03-07:
@GitIssueBot, I think I reported it detailed enough ;-)

fieldOfView, 2020-03-07:
You didn't; there's an issue template which you have to use.

This sounds like a duplicate of https://github.com/foosel/OctoPrint/issues/1407#issuecomment-398277775

frankhommers, 2020-03-09:
I'll reopen it with a detailed report then.

closed 2020-03-09 by frankhommers