OctoPrint/OctoPrint #3477: Updated to 1.4.0 and I cannot login - endless connecting to server screen

NerdboyQ, 2020-03-07:
I have been waiting to update from 1.3 to 1.4 so I have had no issues previously, but once I updated to 1.4 via the octoprint browser I get the screen below. I am never able to login (it's been over 20min).

See my log Here:
[octoprint.log](https://github.com/foosel/OctoPrint/files/4302444/octoprint.log)

Screenshot:
[image]

I have also tried starting in safemode as well, and it does the same thing:
[image]

tedder, 2020-03-08:
There are **no indications in the log that you started in safe mode**. There are indications of what is to blame, but safe mode will show more info. What command did you run to start in safe mode?

NerdboyQ, 2020-03-08:
> There are **no indications in the log that you started in safe mode**. There are indications of what is to blame, but safe mode will show more info. What command did you run to start in safe mode?

<i>octoprint safemode</i>

I just tried a gain today, and now it show up.

NerdboyQ, 2020-03-08:
It's working now after starting in safe mode, then restarting the server with:

<i>sudo service octoprint restart</i>.

closed 2020-03-08 by NerdboyQ