OctoPrint/OctoPrint #3479: Ability to disable M27 polling altogether

frankhommers, 2020-03-09:
#### What were you doing?

1) Connect OctoPrint to my ultimaker.
2) Using my UltiMaker 2 to select a file with the rotary wheel.
3) On first file on which I don't confirm the printer (so I can't reach the file I actually want).
4) OctoPi starts polling with M27 messages.
5) The display shows "Printing with USB ..."

#### What did you expect to happen?

I was hoping to be able to select the file I want at step 3.

#### What happened instead?

The UM2 display shows "Printing with USB ..." all though it isn't printing, but it also disallows access to the rotary wheel.

#### Did the same happen when running OctoPrint in safe mode?

Didn't test this.

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

OctoPi 0.17

#### Printer model & used firmware incl. version

Ultimaker 2. Latest firmware version (as of 2020-03-09)

#### Browser and version of browser, operating system running browser

Chrome 80

#### Link to octoprint.log

Need to capture this if necessary.

#### Link to contents of terminal tab or serial.log

Need to capture this if necessary.

#### Link to contents of Javascript console in the browser

Need to capture this if necessary.

#### Solution:

If I comment out the code which sends M27 in the function _poll_sd_status in comm.py, I can use everything.

What I would really like is the option to disable M27 polling in the configuration/GUI all together.

This is pretty much related to: https://github.com/foosel/OctoPrint/issues/1407#issuecomment-398277775

foosel, 2020-03-09:
Please provide both requested logs.

foosel, 2020-03-23:
@frankhommers any update on the requested logs? I can't look into this without logs and will have to close it.

frankhommers, 2020-03-24:
I haven't been able to produce logs yet.

But the same thing happens as in #1407 

As soon as I rotate the wheel and scroll over a file without selecting it, it sends this:
```
Recv: File opened: BIGCAR~1.GCO Size: 7041501
Recv: File selected
```
That triggers M27 polling to start. And then the display shows "Printing with USB..." and I can't select another file anymore.

foosel, 2020-03-24:
The Ultimaker Marlin build really should finally stop doing this stupid "select on scrolling over" thing.

The thing is, I *have* to start polling if I want to be able to detect prints initiated from the LCD controller (file selection is the only notification I get about this without having to actively query for the SD state). Which is something that people screamed at me to finally implement for years. Of course it relies on a printer actually only a) selecting a file when it is actually about to print it, not just when scrolling through and b) the printer not completely locking up controls for the user just because it receives an M27 on serial (I have to say I don't even understand the logic behind that, an M27 is pretty good indicator that there is in fact NO usb printing going on but rather control is left to the firmware). 

Ah well... since it sounds like the usual Ultimaker shenanigans that I have to admit I thought had been patched by now I'll turn this into a request and look into adding an option for these kind of situations to disable SD printing detection.

frankhommers, 2020-03-24:
Yeah, I agree it's a Ultimaker bug. It shouldn't send selected when it's not finally selected.
But I have to live with it :|

For me it works by commenting out the implementation of _poll_sd_status in comm.py. So it would indeed be very useful to have this behind a feature toggle.

foosel, 2020-04-03:
[image]

Implemented by the above commits. Also automatically applied if Ultimaker 2/2+ firmware is detected. To be released with 1.4.1

frankhommers, 2020-04-10:
That's awesome!!!

frankhommers, 2020-06-02:
I wanted to test this out, but the devel branch leads to a non working situation. And the master branch does not have this yet.

foosel, 2020-06-02:
`mainteinance` branch. Not `devel`. As I said, this is targeted for 1.4.1 (`maintenance`), not 1.5.0 (`devel`).

foosel, 2020-06-02:
For the record, it's documented here: https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#what-do-the-branches-mean

commit 8ee6b9fd (2020-04-03): ✨ Add setting to disable SD print detection

Some firmware variants (Ultimaker 2/2+, maybe others) send a file
selected message while just scrolling through the SD file list on the
integrated controller unit. M27 polling right after file selection might
cause issues for them. This patch allows to work around that.

Closes #3479

See also #1407
closed 2020-08-04 by foosel