OctoPrint/OctoPrint #3482: Can't pass LOGIN screen on a FRAMESET environment

bolsoncerrado, 2020-03-12:
#### What were you doing?

Trying to load several OctoPrints from different RasPis on a <FRAMESET> .html document.

Something as:

`
<frameset rows="40,50%,50%">
<frameset border=0 noresize=yes COLS="100%">
<frame src="buttons.html"  name="[CONTROLS]" scrolling=yes>
</frameset>
<frameset border=0 noresize=yes COLS="*,*,*,*">
<frame src="http://192.168.1.100"  name="[TARS]" scrolling=yes>
<frame src="http://192.168.1.102"  name="[CASE]" scrolling=yes>
<frame src="http://192.168.1.104"  name="[KIPP]" scrolling=yes>
<frame src="http://192.168.1.106"  name="[GERTY]" scrolling=yes>
</frameset>
<frameset border=0 noresize=yes COLS="*,*,*,*">
<frame src="http://192.168.1.108" name="[HAL]" scrolling=yes>
<frame src="http://192.168.1.110" name="[JARVIS]" scrolling=yes>
<frame src="http://192.168.1.112" name="[WOPR]" scrolling=yes>
<frame src="http://192.168.1.114"	name="[VIKY]" scrolling=yes>

</frameset>
</frameset>
`

#### What did you expect to happen?

Get each frame content load each Octoprint instance.

#### What happened instead?

Got the PLEASE LOGIN screen in each instance, and no matter what do we type or click empty, there's no way to get rid of it. 

**However** loading each IP on a clean browser tab works fine.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.0

It was working fine with 1.3.xx

#### Operating System running OctoPrint

OctoPI 0.15.0

#### Printer model & used firmware incl. version

Prusa MK3S on all of them

#### Browser and version of browser, operating system running browser

Any MacOS Browser, latest versions (Safari, Chrome, Firefox)

#### Screenshot(s)/video(s) showing the problem:

https://imgur.com/jm32125

I have read the FAQ.

foosel, 2020-03-12:
Have you disabled frame protection as described here: https://community.octoprint.org/t/my-touchui-display-is-no-longer-working-since-updating-to-octoprint-1-4-0/16819

If not, please try that, it could be related.

bolsoncerrado, 2020-03-12:
Yes I did.

foosel, 2020-03-12:
Can you share your exact frameset file? Via email to gina -at- octoprint.org if it contains sensitive information. I want to make sure I reproduce correctly.

bolsoncerrado, 2020-03-12:
That's exactly the contents of the file. I only call an extra file called buttons.html which only draw two Form buttons to reload the whole thing or call the webcam url tabs, that's all.

foosel, 2020-03-12:
No html tags or anything, just the frameset? So you host it on a server or do you have it lying locally?

lackdaz, 2020-03-13:
I actually have an issue with not being able to get past the login page too on a port forwarded ip address since two weeks ago. I've cleared cache and ran on incognito to no avail. I've noticed this problem on my MacOSX 10.14 browser and android phone. Connecting to the local ip address when on the network works fine.

This is with regards to Version 1.3.10.

Note that I've had these octoprint instances running for 2 years now, and this is a similar problem on two different octoprint instances on two separate networks.

However my windows laptop isn't experiencing said issues and connects fine to the public port.

bolsoncerrado, 2020-03-13:
Sounds like a total different scenario to me, I'd suggest opening a different ticket hmmmm

bolsoncerrado, 2020-03-13:
Did you get my email about it @foosel ? Just checking.

bolsoncerrado, 2020-03-13:
> No html tags or anything, just the frameset? Do you host it on a server or do you have it lying locally?

Did you get my email about it @foosel ? Just checking.

foosel, 2020-03-13:
Nope, I didn't.

bolsoncerrado, 2020-03-13:
Damn. Lmk if you get it on Gmail now.

bolsoncerrado, 2020-03-15:
Did you, now @foosel ?

foosel, 2020-03-16:
Nope. Nothing in the spam folder either. Ah well, I'm back in the office after being sick (ONLY A COLD!) last week and will try my best to repro with what I have.

bolsoncerrado, 2020-03-16:
[image]

Final test going out in 3, 2, 1....

bolsoncerrado, 2020-03-16:
Any luck now @foosel lol

foosel, 2020-03-16:
Yes, that one arrived to the octoprint.org address, I replied to you. 

Meanwhile I've been trying to reproduce but my browsers simply refuse to even render a `frameset`, regardless of what I put in there. A test with an `iframe` (as recommended for modern pages) works just fine and as expected.

bolsoncerrado, 2020-03-16:
I may have to jump to iframe calls then...

Idk what changed tho.

foosel, 2020-03-16:
I just copied your HTML code (thanks for the link) and adjusted it to use local instances in your frames. 

[image]

It's working fine. I can't reproduce any kinds of issues. That is of course as long as I disable the clickjacking protection as outlined [here](https://community.octoprint.org/t/my-touchui-display-is-no-longer-working-since-updating-to-octoprint-1-4-0/16819).

Any errors in the JS console? What exactly happens when you do try to login into one, can you maybe create a video and share it?

bolsoncerrado, 2020-03-16:
I will check the console.

Is there any "status", "check" or report line I can use to call for the variable and see if I applied it already or not? I swear I did the clickjacking disably on every instance already, but just to confirm before commiting twice.

foosel, 2020-03-16:
You can check in `~/.octoprint/config.yaml`. Actually, 

    grep allowFraming ~/.octoprint/config.yaml

should return a line set to true if it worked.

Based on your symptoms I don't think it's that problem though because then it shouldn't even load the login I think. But I might be mistaken. Frankly, I would try if it works with iframes and if does switch to those since that would rule out it being done weirdness caused by the long deprecated frameset stuff.

bolsoncerrado, 2020-03-16:
Yep, I can ack true was returned.

Lemme check about cookies and such...

foosel, 2020-03-17:
Sorry, wrong button

bolsoncerrado, 2020-03-25:
@foosel ,

Today I was re-playing with logins and frames and I was able to catch this on the fly (I NEVER saw this error before with all the frameset tests I did before and during the report of this issue in github). Perhaps some light to the issue?

[image]

bolsoncerrado, 2020-03-25:
[image]

foosel, 2020-04-07:
Thinking it might be [Chrome's SameSite updates](https://www.chromium.org/updates/same-site) ... their limited roll-out would explain why I'm not able to reproduce this. Will do some further tests.

If this is indeed the issue than there's probably nothing I can do about since if I understand it correctly in order for such a setup to get working again you'd need to set both a header AND deliver stuff via https, which due to [ongoing https challenges in LAN setups](https://twitter.com/foosel/status/1022030752349913088) is not possible to default to for OctoPrint. Which would boil down to "Chrome just broken the LAN based web once again, hooray".

bolsoncerrado, 2020-04-08:
Guess what?

"April 3, 2020: In light of the extraordinary global circumstances due to COVID-19, we’ve decided to temporarily roll back the enforcement of SameSite cookie labeling on Chrome 80 stable. We recognize the efforts of sites and individual developers who prepared for this change as part of our ongoing effort to improve privacy and security across the web. We appreciate the feedback from across the web ecosystem which has helped inform this decision. We will provide advance notice on here and the Chromium blog when we plan to resume the gradual rollout, which we’re now aiming for over the summer."

Just tested it back, and it works again! LOL

foosel, 2020-04-15:
I just added a quick option to manually set `SameSite` to `Lax` to replicate the default behaviour that Chrome pushed out earlier, and now I can reproduce. With `Lax`, the login cookie set by the frame doesn't get sent by the browser on reload inside the frame, thus showing the login again/keeping the user logged out.

As far as I understand `SameSite` though, I cannot do anything about this apart from allowing to make `SameSite` and `Secure` configurable so that people like you desiring to run their stuff inside frames can setup https termination through a reverse proxy, then set `SameSite` to `None` and `Secure` to `true`. https termination is required since according to [this](https://www.chromestatus.com/feature/5633521622188032) browsers should only support `SameSite=None` if cookies are set in a secure context/through an https connection. So, once again, browser vendors have introduced a feature to make browsing safer (good) and completely forgotten about LAN based sites that do not have domains, certificates and in general no administration staff (bad). Yay. [And this keeps happening and there's nothing developers like myself can do about it](https://twitter.com/foosel/status/1022030752349913088).

I've pushed the new settings for cookie configuration and also added docs about those. That's all I can do here. Things will break again once `SameSite=Lax` gets rolled out again, so you'll either need to change your setup to https or find a way for your browser to not enforce `Lax`/allow `None` even outside of a `Secure` context. My hands are tied here.

bolsoncerrado, 2020-04-15:
Well, thanks for that!

bolsoncerrado, 2020-07-31:
Howdy!

Seems they rolled out SameSite again.

Just to confirm <iframe> is failing to operate also as much as <frameset> did and wanted to confirm nothing can be done to fool the system somehow?

Am I really the only being in the planet that needs to monitor more than 2 printers running octoprint without willing to switch into multiple windows, tabs or whatever remotely, every time??

foosel, 2020-07-31:
Nothing can be done.

closed 2020-07-31 by foosel

jneilliii, 2020-07-31:
OctoFarm

bolsoncerrado, 2020-07-31:
I was able to still use frames ets with Opera browser... But that Octofarm thing looks interesting, thanks!

fcapri, 2021-03-18:
i'm having the same problem. i've been using multiple windows of chrome next to eachtother but i wanted to put them in 1 screen
[image]

been using frameset, i got the "octopi refused connection errors". i 've disabled frame protection and i've got the login screen 2 times. mine also worked with frame but can't get the login page.
[image]

i then made the same html file as above with the iframes, but here still can't get past the login screen. 
[image]

any suggestions what i still can do?

bolsoncerrado, 2021-03-18:
I use Opera 70.0.3728.95 to still see all my printers using framesets. Any version past that will render Login screens.
I keep Opera as "Farm Software" and Chrome as my regular internet browser, among others, that keep auto-updating as time goes by, but I disabled Autoupdate on Opera for the frameset reason.

foosel, 2021-03-18:
As I already explained, this is out of my hands.

commit 00a5243e (2020-04-15): ✨ config flag to configure SameSite and Secure cookie headers

Provides one part of the solution to #3482. The other part is sadly
on the shoulders of the user, who needs to setup either a whole domain
hierarchy to not run into "third party page" issues that would trigger
SameSite=Lax blocking, or needs to reconfigure their setup to use
https and make sure to set Secure to true.

Closes #3482 (there's nothing else I can do here)