OctoPrint/OctoPrint #3483: [Request] Additional Events - Start Up and Shut Down

JHale716, 2020-03-14:
With my installation of OctoPi I'm using WiFi switches and IFTTT for power control. Less messing around and they were spare and handy.

I've found with the standard Startup/Shutdown events that they can create issues if they are linked to IFTTT. i.e software update that turned off the Pi before it was done with the shutdown routine and toasted my install.

Additional Events that would be useful:
- Restarting, enabling a trigger for say switching off the external webcam to reboot it at the point of start-up, as Shutdown would have given it the power off prompt needed.
- OctoPiOn, issued as soon as possible to power on additional peripherals that are needed, ie. external light maybe, again on a WiFi switch. 
- OctoPiOff, issued when safe to power off

Startup as an indicator of up; and
Shutdown as an indicator of the start of the shutdown process are fine

So the use case:
Startup: Could be used for lights and some peripherals
Shutdown: Powers off webcam (Could be used for lights too)
Restarting: Powers on Webcam
OctopiOn: Powers on Webcam, Lights, and other peripherals
OctopiOff: Powers off lights and other peripherals
 
Does that make sense? 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Floppy, 2020-04-12:
This would be great. I'd love toe be able to send some GCODE before server shutdown, to set a message on the printer LCD to tell me that I'm safe to power off after a few seconds.

MrSlimbrowser, 2022-01-13:
I'd additionally love to see a "BeforeConnecting" event as well.
UseCase: Turning on the socket the printer is plugged into when hitting the connect button (e.g. issuing "curl http://192.168.2.241/cm?cmnd=Power%20On" when using a tasmota controlled socket).

foosel, 2022-01-13:
Events are processed asynchronously, so while it would be possible to add such an event, it would be pretty useless since you can't use it to do some blocking "do this first" kinda thing. A plugin hook would make more sense here.

jneilliii, 2022-01-13:
> I'd additionally love to see a "BeforeConnecting" event as well. UseCase: Turning on the socket the printer is plugged into when hitting the connect button (e.g. issuing "curl http://192.168.2.241/cm?cmnd=Power%20On" when using a tasmota controlled socket).

You should use either my tasmota plugin or psucontrol with the tasmota subplugin. Both handle connecting for you after powering on.