OctoPrint/OctoPrint #3484: Updating OctoPrint from 1.3.10 to 1.4 fails "ImportError: No module named extern"

battlezone, 2020-03-14:
#### What were you doing?

1. Update OctoPrint via settings -> Software update

#### What did you expect to happen?
I expected OctoPrint to update succesfully,
#### What happened instead?
OctoPrint failed to update.
#### Did the same happen when running OctoPrint in safe mode?
Same thing happens in safe mode.

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
P3Steel running on MKS Sbase, but N/A since not connected

#### Browser and version of browser, operating system running browser
Vivaldi 2.5.1525.40 on Windows 10

#### Link to octoprint.log
Octoprint.log https://pastebin.com/8fBdsi4U
Plugin_softwareupdate_console.log: https://pastebin.com/63L6Y9q8

#### Link to contents of terminal tab or serial.log
Serial log not enabled, terminal is empty

#### Link to contents of Javascript console in the browser
update console contains same information as Plugin_softwareupdate_console.log: https://pastebin.com/63L6Y9q8

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

inajob, 2020-03-21:
My OctoPrint also occurred same error. My OctPrint is version 1.3.11.

sam-jeffs, 2020-03-21:
Also having the same issue, from 1.3.11.

Phr3d13, 2020-03-26:
Also having this issue. from 1.3.11 to 1.4.0

foosel, 2020-03-26:
I need logs and environment information from all of you. If you are all on OctoPi 0.14 I suggest you make a backup and flash to a current version of OctoPi, I no longer support 0.14, it's simply too ancient by now.

Based on a quick google search it might be that `~/oprint/bin/pip install -U setuptools` via SSH fixes whatever is broken here, but I don't know. The thing is, the issue arises in a third party dependency of OctoPrint, not OctoPrint itself, and on the officially supported environments (OctoPi 0.15 through 0.17, so all OctoPi releases of the past 2 years) I cannot reproduce this.

inajob, 2020-03-26:
Oh, My environment is OctoPi 0.14.0. I'll upgrade. thank you @foosel !

inajob, 2020-03-26:
@foosel I know only the way that re-install OctoPi. Is there the way to upgrade OctPi?

foosel, 2020-03-26:
No, not really. Too many variables outside of our control. You might have success with an apt-get upgrade and a full rebuild of the virtual environment, but at that point you have to do the same effort of backup/restore that you'd have to do with a fresh flash anyhow. 

Create a backup, flash a fresh OctoPi 0.17 on a card, run through first setup (wifi + wizard) and then restore your backup.

capnkrunchy, 2020-03-29:
I was having this same issue attempting to upgrade to 1.4.0 from 1.3.10.

I fixed the issue by running the following commands:
`sudo su
source OctoPrint/venv/bin/activate
pip install --upgrade --force-reinstall setuptools`
and attempting to upgrade via the web interface again.

foosel, 2020-03-29:
You definitely do not need the `sudo su` there and it will only cause issues down the road since you now modified files as super user which might no longer be modifiable as regular user due to that. Best run `chmod -r pi.pi ~/oprint` to fix the permissions again.

Activate and upgrade of setuptools *as regular user* should suffice.

I'm also closing this since everything points towards this issue being caused by outdated environments and that is outside of the control of OctoPrint.

closed 2020-03-29 by foosel

mantisrobot, 2020-04-04:
I had the same issue upgrading from V1.3.9 to V1.4.0
This fixed install errors for me, from the terminal:
/home/pi/oprint/bin/python -m pip install --upgrade --force-reinstall setuptools

54696d21, 2020-04-10:
I had to do to fix it
`/home/pi/oprint/bin/python -m pip install --upgrade --force-reinstall pip`

u173487, 2020-05-22:
I had the same problem

/home/pi/oprint/bin/python -m pip install --upgrade --force-reinstall pip 

Worked for me too. I suspect pip was not updating correctly until i ran this command.

```
pi@octopi:~/.octoprint/logs $ /home/pi/oprint/bin/python -m pip install --upgrade --force-reinstall pip
^COperation cancelled by user
You are using pip version 9.0.1, however version 20.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
pi@octopi:~/.octoprint/logs $ sudo !!
sudo /home/pi/oprint/bin/python -m pip install --upgrade --force-reinstall pip
Collecting pip
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/43/84/23ed6a1796480a6f1a2d38f2802901d078266bda38388954d01d3f2e821d/pip-20.1.1-py2.py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 168kB/s
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Uninstalling pip-9.0.1:
      Successfully uninstalled pip-9.0.1
Successfully installed pip-20.1.1

Pip refused to upgrade from V9.0.1 to 22.1.1 until i ran this command