OctoPrint/OctoPrint #3485: Octoprint disconnects when print is finished

JWolfJr, 2020-03-15:
Today I just finished up a print. After the print finished I get a M104 (set hotend temp), then M117 (for sending info to lcd screen), finally a M400 (Marlin docs: "This command causes all GCode processing to pause and wait in a loop until all moves in the planner are completed."). Then octoprint counts down and eventually times out and disconnects. M400 and M117 are in my long running commands.  If it helps, my setup as follows:
- Ender-3
-SKR mini E3 V1.2
-BTT TFT35 V3 touch screen
-Raspberrypi 3B+ running latest octoprint
-Marling 2.0.5 latest build

This did just start happening after reflashing the SKR mini to the latest build (2.0.5), and I was not having this issue with 2.0.4.  I've been looking through the docs and compared both builds, I'm unable to find the difference to cause this.  Is anyone experiencing the same?

foosel, 2020-03-15:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2020-03-15 by foosel

JWolfJr, 2020-03-16:
My apologize, thank you!